gnuradio/gnuradio #2041: Drop-down menus on GRC toolbar have poor contrast

argilo, 2018-09-24:
On the master branch, the drop-down menus in GRC to create a new flowgraph or open a recent one are displayed with grey text on a grey background:

[image]

[image]

This gives the impression that the menu items are disabled.

btashton, 2018-11-03:
@argilo are you still seeing this on master?  Building on Fedora 29 I they have the proper black text, might be a GTK theme issue?

argilo, 2018-11-04:
@btashton Indeed, the problem is gone on the latest master. I now see black text instead of grey. I'll close this off.

closed 2018-11-04 by argilo