gnuradio/gnuradio #2049: Virtual Sink and Virtual Source are broken

argilo, 2018-09-26:
On the master branch, Virtual Sink and Virtual Source (from the "Misc" category) are broken. To reproduce the problem, drop either block onto a flowgraph in GRC, and an error message will appear in the console. For Virtual Sink I get the following:
```
Traceback (most recent call last):
  File "/home/argilo/prefix_next/lib/python3.6/dist-packages/gnuradio/grc/gui/DrawingArea.py", line 127, in _handle_drag_data_received
    self._flow_graph.add_new_block(selection_data.get_text(), coords)
  File "/home/argilo/prefix_next/lib/python3.6/dist-packages/gnuradio/grc/gui/canvas/flowgraph.py", line 147, in add_new_block
    block = self.new_block(key)
  File "/home/argilo/prefix_next/lib/python3.6/dist-packages/gnuradio/grc/core/FlowGraph.py", line 253, in new_block
    block = self.parent_platform.make_block(self, block_id, **kwargs)
  File "/home/argilo/prefix_next/lib/python3.6/dist-packages/gnuradio/grc/core/platform.py", line 413, in make_block
    return cls(parent, **kwargs)
  File "/home/argilo/prefix_next/lib/python3.6/dist-packages/gnuradio/grc/gui/canvas/block.py", line 46, in __init__
    super(self.__class__, self).__init__(parent, **n)
  File "/home/argilo/prefix_next/lib/python3.6/dist-packages/gnuradio/grc/core/blocks/virtual.py", line 44, in __init__
    super(VirtualSink, self).__init__(parent, **kwargs)
  File "/home/argilo/prefix_next/lib/python3.6/dist-packages/gnuradio/grc/core/blocks/block.py", line 79, in __init__
    self.sinks = [port_factory(parent=self, **params) for params in self.inputs_data]
  File "/home/argilo/prefix_next/lib/python3.6/dist-packages/gnuradio/grc/core/blocks/block.py", line 79, in <listcomp>
    self.sinks = [port_factory(parent=self, **params) for params in self.inputs_data]
  File "/home/argilo/prefix_next/lib/python3.6/dist-packages/gnuradio/grc/core/platform.py", line 421, in make_port
    return cls(parent, **kwargs)
TypeError: __init__() missing 1 required positional argument: 'direction'
```

haakov, 2018-10-01:
Yep, these are indeed broken. I did have a PR for this a couple of months ago, but I believe it stopped working after `python3` was merged into `next`. I'll have a look at it

marcusmueller, 2018-10-27:
Said PR #2102 has been merged. @argilo, does that fix the bug for you, too? (if you can, close please!)