gnuradio/gnuradio #2052: Default Header Format Obj. is broken

argilo, 2018-09-26:
On the master branch, adding a Default Header Format Obj. block (from the "Packet Operators" category) to a flowgraph causes things to break. If its properties dialog is opened an closed, or if another block is added to the flowgraph, the following error occurs:
```
Traceback (most recent call last):
  File "/home/argilo/prefix_next/lib/python3.6/dist-packages/gnuradio/grc/core/blocks/_templates.py", line 75, in render
    return template.render(**namespace)
  File "/usr/lib/python3/dist-packages/mako/template.py", line 462, in render
    return runtime._render(self, self.callable_, args, data)
  File "/usr/lib/python3/dist-packages/mako/runtime.py", line 838, in _render
    **_kwargs_for_callable(callable_, data))
  File "/usr/lib/python3/dist-packages/mako/runtime.py", line 873, in _render_context
    _exec_template(inherit, lclcontext, args=args, kwargs=kwargs)
  File "/usr/lib/python3/dist-packages/mako/runtime.py", line 899, in _exec_template
    callable_(context, *args, **kwargs)
  File "memory:0x7fded1f87b70", line 25, in render_body
ValueError: invalid literal for int() with base 10: "'0'"

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/argilo/prefix_next/lib/python3.6/dist-packages/gnuradio/grc/gui/Application.py", line 548, in _handle_action
    flow_graph.import_data(n)
  File "/home/argilo/prefix_next/lib/python3.6/dist-packages/gnuradio/grc/core/FlowGraph.py", line 370, in import_data
    self.rewrite()  # evaluate stuff like nports before adding connections
  File "/home/argilo/prefix_next/lib/python3.6/dist-packages/gnuradio/grc/core/FlowGraph.py", line 181, in rewrite
    self.renew_namespace()
  File "/home/argilo/prefix_next/lib/python3.6/dist-packages/gnuradio/grc/core/FlowGraph.py", line 212, in renew_namespace
    for variable_block in self.get_variables():
  File "/home/argilo/prefix_next/lib/python3.6/dist-packages/gnuradio/grc/core/FlowGraph.py", line 79, in get_variables
    return expr_utils.sort_objects(variables, attrgetter('name'), methodcaller('get_var_make'))
  File "/home/argilo/prefix_next/lib/python3.6/dist-packages/gnuradio/grc/core/utils/expr_utils.py", line 74, in sort_objects
    id2expr = {get_id(obj): get_expr(obj) for obj in objects}
  File "/home/argilo/prefix_next/lib/python3.6/dist-packages/gnuradio/grc/core/utils/expr_utils.py", line 74, in <dictcomp>
    id2expr = {get_id(obj): get_expr(obj) for obj in objects}
  File "/home/argilo/prefix_next/lib/python3.6/dist-packages/gnuradio/grc/core/blocks/block.py", line 228, in get_var_make
    return self.templates.render('var_make')
  File "/home/argilo/prefix_next/lib/python3.6/dist-packages/gnuradio/grc/core/blocks/_templates.py", line 77, in render
    raise TemplateError(error, text)
gnuradio.grc.core.errors.TemplateError: (ValueError('invalid literal for int() with base 10: "\'0\'"',), '% if int(access_code)==0:\nself.${id} = ${id} = digital.header_format_default(digital.packet_utils.default_access_code,\\\n${threshold}, ${bps})\n% else:\nself.${id} = ${id} = digital.header_format_default(${access_code},\\\n${threshold}, ${bps})\n% endif')
```

argilo, 2018-10-02:
i dug into this a bit, and it seems there are a couple problems. First, when the block's XML was converted to YML, `access_code()` in the if statement was converted to `access_code`:

https://github.com/gnuradio/gnuradio/blob/007923c64d2505b37dffb8e11cb1e3ecf4e489eb/gr-digital/grc/variable_header_format_default.block.yml#L22

But `access_code()` would need to be used to access the underlying Python object (rather than its string representation):

https://github.com/gnuradio/gnuradio/blob/007923c64d2505b37dffb8e11cb1e3ecf4e489eb/grc/core/params/template_arg.py#L21-L27

Unfortunately, adding back the brackets still leaves things broken. It seems that sometimes GRC doesn't pass in a properly functioning `TemplateArg` object, and `access_code()` returns `None`.

@haakov Maybe you have some idea what's broken here.

haakov, 2018-10-22:
Hmm I'm a bit confused by this, are you sure adding the parentheses back doesn't work? I just tried it, looks like it worked

haakov, 2018-12-18:
Would love a second (third) opinion on this.

IIRC, the proposed fix is changing:

https://github.com/gnuradio/gnuradio/blob/007923c64d2505b37dffb8e11cb1e3ecf4e489eb/gr-digital/grc/variable_header_format_default.block.yml#L22

To `% if int(access_code())==0:`

mormj, 2019-05-28:
The parenthesis didn't fix it - calling the access_code() accessor returns NoneType, so fails with a different mode.  Extra quotes are being added to the template arguments as in several other issues, but this block is unique because the input to grc is string, but input to underlying make function is uint64.  

see PR #2516

PR #2516 (2019-05-29): digital: default header format object template issues
Since the first argument of the header_format_default object
is a uint64 but the template argument is a string, there is
some confusion and addition of unnecessary quotes

Fix by stripping the quotes with eval() in the var_make

Fixes #2052
commit cea8c170 (2019-05-29): digital: default header format object template issues

Since the first argument of the header_format_default object
is a uint64 but the template argument is a string, there is
some confusion and addition of unnecessary quotes

Fix by stripping the quotes with eval() in the var_make

Fixes #2052
closed 2019-05-29 by mbr0wn