gnuradio/gnuradio #2053: Filter Design Tool doesn't run

argilo, 2018-09-27:
On the master branch, the GNU Radio Filter Design Tool (gr_filter_design) no longer works. It runs in Python 3, but it depends on PyQwt, which does not support Python 3.

The PyQwt project appears to be dead, so it seems the only path forward is to remove that dependency. PythonQwt could be a suitable replacement.

mormj, 2019-05-16:
PyQtGraph looks doable - Lots of changes in PyQt4 in going to python3 as well

mormj, 2019-05-21:
Picked this up to where it runs, but there is still some missing/different functionality:

https://github.com/gnuradio/gnuradio/pull/2495

mormj, 2019-05-21:
There are remaining missing functionalities and bugs, namely:

- Sizing is different - minimum size is larger than the previous version
- Pole Zero plot is not interactive
- Buffer current plot is disabled
- Ideal band plot is disabled
- Plot Zoom is not quite right, or I haven't figured out the default
behavior
- Axes in overlay mode need to be colored to match the curves
- Second plot on overlay goes outside the axes - apparently it has a
larger bounding box

Looking for feedback on prioritization on these issues

bhilburn, 2019-05-21:
Wow! A wild @mormj appears and totally rescues this work! Nice job, Josh!

My feedback on priorities of those items:

Priorities worth getting done for v3.8:

1. Plot Zoom is not quite right, or I haven't figured out the default
    behavior
2. Second plot on overlay goes outside the axes - apparently it has a
    larger bounding box

Mark as TODOs (open enhancement issues):

1. Ideal band plot is disabled
2. Buffer current plot is disabled
3. Pole Zero plot is not interactive
4. Axes in overlay mode need to be colored to match the curves

Mark as wont-fix:

1. Sizing is different - minimum size is larger than the previous version

Thoughts, @marcusmueller ?

mormj, 2019-07-11:
The blocking issue of the basic functionality for filter design tool has been merged as:
#2495 
The remaining issues as noted above have been added as a separate issue.  I believe this can be closed @marcusmueller

closed 2019-07-13 by marcusmueller