gnuradio/gnuradio #2055: Tag propagation doesn't work correctly when relative rate changes in runtime

ptrkrysik, 2018-09-30:
The 'set_relative_rate' makes misleading impression that GNU Radio blocks have support for runtime changes of processing block's relative rate (number of output samples / number of input samples).

However offset of tags is incorrectly computed from the moment when 'set_relative_rate' function changes the relative rate for the first time in the runtime. The reason for that is incorrect computation of output tag offset in 'propagate_tags' function:
https://github.com/gnuradio/gnuradio/blob/007923c64d2505b37dffb8e11cb1e3ecf4e489eb/gnuradio-runtime/lib/block_executor.cc#L140
https://github.com/gnuradio/gnuradio/blob/007923c64d2505b37dffb8e11cb1e3ecf4e489eb/gnuradio-runtime/lib/block_executor.cc#L149
https://github.com/gnuradio/gnuradio/blob/007923c64d2505b37dffb8e11cb1e3ecf4e489eb/gnuradio-runtime/lib/block_executor.cc#L183
https://github.com/gnuradio/gnuradio/blob/007923c64d2505b37dffb8e11cb1e3ecf4e489eb/gnuradio-runtime/lib/block_executor.cc#L191

Tag offset of the tag at the output is computed using this equation:
y = a*x
where:
* 'a' is relative rate
* 'x' is the offset of the tag at the input
* 'y' is the offset of the tag at the output

This assumes that from the beginning relative rate was equal to 'a', which isn't true after the relative rate is changed.

Why this is wrong? Look at the images below what happens on increasing and decreasing relative rate.
[image]
[image]

To fix this the information when the relative rate was changed (a pair of input sample number 'n' and output samples number 'm') can be used:
y = a*(x-n)+m

I don't know yet how to pass the information about '(n,m)' pair to the 'set_relative_rate' fuction.
It would be perfect if it could get it from 'nitems_read(...)' and 'nitems_written(...)' functions.

Also when changing the relative rate all tags with offsets before rate change should be propagated with use of the old rate to compute their offsets. To assure that the 'set_relative_rate' could call 'propagate_tags' function for all tags with offset equal or less than 'n'.

I will also later show what is wrong in an example based on 'mmse_resampler', that allows runtime relative rate changes (I will edit this issue and add a link to it). There are other issues with tag propagation in that block, hence there will be a separate issue.

awalls-cx18, 2019-01-17:
I thought about this as well about a year ago.  (With exactly the same sort of graphs in my head as @ptrkrysik has actually plotted here).
See this email thread.
https://lists.gnu.org/archive/html/discuss-gnuradio/2018-01/msg00042.html

marcusmueller, 2019-04-21:
I believe we have rational tag offsets now. How much of this issue is addressed by these?

ptrkrysik, 2019-04-22:
The more precise positions of tags after resampling is nice feature but
it doesn't fix the issue with wrong computation of tags positions after
relative rate of a resampler is changed dynamically in the runtime. I
looked at set_relative_rate and it seems that that it still doesn't
store moment (input and output sample number) of relative rate change.
Which is what I think prerequisite to implement dynamic rate changes.

I can try to prepare some pull request when I'll have a bit more time.

noc0lour, 2019-05-14:
I'm going to hack on this during the hackfest at the ATA now.

noc0lour, 2019-05-15:
@ptrkrysik Comments on #2484 are definitely needed and appreciated since I'm not super sure this works as intended. If you can come up with a short test case which fails with current behaviour and should pass with future behaviour that would be great. 

A caveat which we can't solve with current architecture: relative rate is the rate for every input to every output.

ptrkrysik, 2019-05-15:
I'll try to prepare something based on the flowgraph that was created in
order to write this issue.

awalls-cx18, 2019-05-15:
[snip]
> A caveat which we can't solve with current architecture: relative rate is the rate for every input to every output.

Right, the API needs to change.

Internally to convert an input tag sample offset to an output tag sample offset, the runtime only uses 4 parameters: an implicit input base offset of 0, an implicit output base offset of 0, the relative rate, and the tag delay for the block.

Ignoring the tag delay for the moment, the runtime now needs to keep track of tuples of (input base offset, output base offset, relative rate) for each change of relative rate, and keep those tuples around until the current and future calls to general_work() no longer need older tuples, which can then be discarded.

For each relative rate change, a block needs to inform the runtime of not only the new relative rate, but also the applicable input base offset and/or output base offset at which the new relative rate applies.  Blocks should be able to do this explicitly.  To avoid changing a lot of existing block code, you may also want to add some logic for blocks to inform the runtime implicitly. (Maybe at the return from the call to work?)

I'll try to look at #2484.

noc0lour, 2019-05-15:
For now I'd like to add the right calculations assuming the rate applies for every combination of inputs and outputs since that change would be to big for the upcoming release. Saving tuples and rate on a change can be done without changing too much of the current API, that's why I proposed that change now in #2484

ptrkrysik, 2019-05-16:
W dniu 15.05.2019 o 05:19, Andrej Rode pisze:
> A caveat which we can't solve with current architecture: relative rate
> is the rate for every input to every output.

I'm aware of that, but this is more limitation of current runtime, not a
bug like incorrect tag positions after set_relative_rate.

ptrkrysik, 2019-05-31:
The test case:
[test_tags_3.8.tar.gz](https://github.com/gnuradio/gnuradio/files/3241688/test_tags_3.8.tar.gz)

There are two test cases that you run with:
test_bad_increasing.py # increasing resampling ratio,
test_bad_decreasing.py # decreasing resampling ratio,

The tests are not prefect as they use mmse_resampler (out of lack of other options in GNU Radio) which itself isn't correct. The most important fix that is needed in respect to this test is that it should call 'set_inverse_relative_rate' for each relative rate change and not just once before finishing 'work' function.

Another fix that is needed (but not crucial to this test) is that it assumes there are enough input samples to produce 'noutput' number of samples. This is not known beforehand as the sample rate might change and more samples might be needed than there are in the input buffers.

ptrkrysik, 2019-05-31:
The most important part of the test are expected_offsets of tags at the output of the resampler. They are compared to obtained tags positions and if they agree the test passes, if not it fails. expected_offsets were obtained from gr-gsm's fractional resampler (they are the same as in #2056).

Possibility of fractional position of tags is not taken into account and gr-gsm's fractional resampler might not give perfect result. Anyway if GNU Radio would give such results it would be big improvement in respect to what there is currently.

noc0lour, 2019-06-06:
@ptrkrysik Thank you for providing the necessary test cases and descriptions. As I'm not in a position to work on this in a timely manner I'm hoping someone else might pick up already started work on this. I referenced a PR here #2484.

marcusmueller, 2023-12-22:
My gut feeling is that this issue is still present, but there's been a bit of  movement on the tag offset front in the last couple years, so looking into this requires actual work

awalls-cx18, 2023-12-22:
It is a non-trivial change to a core part of GNURadio.  I doubt it being fixed would go unnoticed.  It is safe to assume the problem still exists.

Joaohsd, 2024-02-23:
I was testing a module that was developed in GNURadio 3.7, which did not have any issue with memory leak. Migrating to 3.8, I saw that there was a memory leak in gr::buffer::add_item_tag method. I got that using heaptrack software. Maybe, it is a problem related and I can open another issue for that, because they are related, but not the same (Maybe). I have a block that changes the relative rate according to the control information received. So, sometimes we do not produce any data in the output or produce some data that does not respect the relative rate defined or produce data in correct way. I solved this memory leak on my block configuring tag propagation policy of this block to TPP_DONT. I think they are problem related and I know that I should not disrespect the relative rate defined, but it was working in older version of GNURadio.