gnuradio/gnuradio #2056: A few issues with tag propagation in mmse_resampler

ptrkrysik, 2018-10-01:
A few issues with tag propagation in mmse_resampler

'mmse_resampler' seems to offer ability to have the relative rate changed in the runtime, as it has 'rate' input and can accept 'resamp_ratio' messages. However this functionality doesn't work correctly with tag propagation - offset of tags at the output are incorrect. One of the reasons is more general issue #2055 in GNU Radio runtime, but the 'mmse_resampler' ads problems of it own on top of it:
* after getting 'resamp_ratio' it sets it's internal settings with 'set_resamp_ratio', but it doesn't call the 'set_inverse_relative_rate' function of the upper class, so not only tags offsets will be computed incorrectly, but also everything that depends on relative rate of this block won't work as it should.
https://github.com/gnuradio/gnuradio/blob/007923c64d2505b37dffb8e11cb1e3ecf4e489eb/gr-filter/lib/mmse_resampler_cc_impl.cc#L71:L77

* it calls 'set_inverse_relative_rate' only for the last change of the signal on the 'rate' input, losing the information about previous rate changes.

https://github.com/gnuradio/gnuradio/blob/007923c64d2505b37dffb8e11cb1e3ecf4e489eb/gr-filter/lib/mmse_resampler_cc_impl.cc#L126:L142

Also, I have a bit of issue with this 'rate' input, as there are two other ways to control blocks settings: messages and tags (for which the precise moment of change is known by definition because they have the offset attached to them). In case nobody is using this 'rate' input, why not drop it and replace with processing of tags. The example how to do it is a 'controlled_fractional_resampler' I implemented based on 'fractional_resampler' from GNU Radio (old name of 'mmse_resampler'):
https://github.com/ptrkrysik/gr-gsm/blob/master/lib/misc_utils/controlled_fractional_resampler_cc_impl.cc

'controlled_fractional_resampler' disables automatic tag propagation and replaces it with it's own to set correct tag offsets. This part of its functionality shouldn't be needed after #2055 issue will be fixed.

I prepared a comparison the results obtained by 'mmse_resampler' and 'controlled_fractional_resampler'. The GRC file that was used to get it is attached to this issue. Tags at the input appear every 3 samples:

```
Tag Debug: input
Input Stream: 00
  Offset: 3  Source: n/a     Key: T   Value: 1
  Offset: 6  Source: n/a     Key: T   Value: 2
  Offset: 9  Source: n/a     Key: T   Value: 3
  Offset: 12  Source: n/a     Key: T   Value: 4
  Offset: 15  Source: n/a     Key: T   Value: 5
  Offset: 18  Source: n/a     Key: T   Value: 6
  Offset: 21  Source: n/a     Key: T   Value: 7
  Offset: 24  Source: n/a     Key: T   Value: 8
  Offset: 27  Source: n/a     Key: T   Value: 9
  Offset: 30  Source: n/a     Key: T   Value: 10
```

At sample '12' resamp  ratio is changed. First lets look what happens at the output of 'controlled_fractional_resampler'  when resamp ratio is decreased from 1.0 to 0.5 (what means the block's relative rate is increased to 2.0), which is what I think is correct way to propagate tags and should be also output 'mmse_resampler':
```
Tag Debug: controlled_fractional_resampler
Input Stream: 00
  Offset: 3  Source: n/a     Key: T   Value: 1
  Offset: 6  Source: n/a     Key: T   Value: 2
  Offset: 9  Source: n/a     Key: T   Value: 3
  Offset: 12  Source: n/a     Key: set_resamp_ratio   Value: 0.5
  Offset: 12  Source: n/a     Key: T   Value: 4
  Offset: 18  Source: n/a     Key: T   Value: 5
  Offset: 24  Source: n/a     Key: T   Value: 6
  Offset: 30  Source: n/a     Key: T   Value: 7
  Offset: 36  Source: n/a     Key: T   Value: 8
  Offset: 42  Source: n/a     Key: T   Value: 9
  Offset: 48  Source: n/a     Key: T   Value: 10
```
[image]

Tags before the change have their offset unchanged and only tags after the change have appear every 6 samples.

In the same situation tags at the output of 'mmse_resampler' have following offsets:
```
Tag Debug: mmse_resampler
Input Stream: 00
  Offset: 6  Source: n/a     Key: T   Value: 1
  Offset: 12  Source: n/a     Key: T   Value: 2
  Offset: 18  Source: n/a     Key: T   Value: 3
  Offset: 24  Source: n/a     Key: T   Value: 4
  Offset: 30  Source: n/a     Key: T   Value: 5
  Offset: 36  Source: n/a     Key: T   Value: 6
  Offset: 42  Source: n/a     Key: T   Value: 7
  Offset: 48  Source: n/a     Key: T   Value: 8
  Offset: 54  Source: n/a     Key: T   Value: 9
  Offset: 60  Source: n/a     Key: T   Value: 10
```
[image]

The block sets the relative rate straight away, and even tags before the change will have their offset changed, so they appear every 6 samples now. This surely is not what user expects. But it's even funnier when the resamp ratio is increased. Let's try to change it from 1.0 to 3.0 and see what will happen. For 'controlled_fractional_resampler' this is the result (again this is what I think is correct):
```
Tag Debug: controlled_fractional_resampler
Input Stream: 00
  Offset: 3  Source: n/a     Key: T   Value: 1
  Offset: 6  Source: n/a     Key: T   Value: 2
  Offset: 9  Source: n/a     Key: T   Value: 3
  Offset: 12  Source: n/a     Key: set_resamp_ratio   Value: 3
  Offset: 12  Source: n/a     Key: T   Value: 4
  Offset: 13  Source: n/a     Key: T   Value: 5
  Offset: 14  Source: n/a     Key: T   Value: 6
  Offset: 15  Source: n/a     Key: T   Value: 7
  Offset: 16  Source: n/a     Key: T   Value: 8
  Offset: 17  Source: n/a     Key: T   Value: 9
  Offset: 18  Source: n/a     Key: T   Value: 10
```
[image]

After the change tags appear every 1 sample. Now 'mmse_resampler':
```
Tag Debug: mmse_resampler
Input Stream: 00
  Offset: 0  Source: n/a     Key: T   Value: 1
  Offset: 0  Source: n/a     Key: T   Value: 2
  Offset: 0  Source: n/a     Key: T   Value: 3
  Offset: 0  Source: n/a     Key: T   Value: 4
  Offset: 0  Source: n/a     Key: T   Value: 5
  Offset: 0  Source: n/a     Key: T   Value: 6
  Offset: 0  Source: n/a     Key: T   Value: 7
  Offset: 0  Source: n/a     Key: T   Value: 8
  Offset: 0  Source: n/a     Key: T   Value: 9
  Offset: 0  Source: n/a     Key: T   Value: 10
```
[image]

All tags have offset 0, which means that tag propagation won't work at all when the resamp ratio is increased ;).

So TL;DR what I think is needed to do:
* fix issue #2055,
* add setting of block's relative rate when it receives a "resamp_ratio" message,
* add support for setting resamp ratio with tags, by using gr-gsm's 'controlled_fractional_resampler' as a starting point,
* preferably remove the 'rate' input as it's a bit nasty way to control the rate.

jaredd, 2021-01-08:
@ptrkrysik This is a great writeup. I was about a quarter of the way through re-writing this EXACT BLOCK because of the issues described when I decided to look up if anyone else did this. I would add some more issues:

1. The Fractional Resampler block should just be renamed MMSE Resampler in GRC to better clue people to the correct c++ files and also better differentiate from the other resamplers (I would also recommend the Fractional Interpolator to be renamed to MMSE Interpolator).
2. In addition to rate changes, the phase should also be tag settable. This is particularly important when using external time estimates, e.g. the corr_est block.

I wanted to ask if there's been any progress on this that I can't find? Otherwise I will try to get this coded up, prob starting with the gr-gsm implementation.

marcusmueller, 2023-12-22:
Related to, and hence in same state as, #2055