gnuradio/gnuradio #2065: codec2/freedv_api.h:64:71: error: ‘size_t’ has not been declared

mayorbobster, 2018-10-04:
Checked all dependencies, gnuradio would not compile.
Fixed by adding:
#include <stdlib.h> 
to gr-vocoder/include/gnuradio/vocoder/api.h
System is Linux Mint 19,

marcusmueller, 2019-04-21:
We've worked quite a bit on our build system. Does this still apply, @mayorbobster?

mayorbobster, 2019-04-23:
Yes.  Deleted old gnuradio git, pulled new one, rebuilt, problems reoccur.  Once again added #include <stdlib.h> to gr-vocoder/include/gnuradio/vocoder/api.h, recompiled successfully.  stdlib.h is included in the gnuradio stack in at least 122 places, but none seem to be within the name binding scope of vocoder/api.h.  I use the vocoder stack to run dsd and other applications.
I normally build gnuradio as part of a stack.  I will post my stack build script and the script I use to build gnuradio (referenced in the first script).
[build.gr.stack.sh.txt](https://github.com/gnuradio/gnuradio/files/3109267/build.gr.stack.sh.txt)
[getgnuradio.sh.txt](https://github.com/gnuradio/gnuradio/files/3109271/getgnuradio.sh.txt)

mayorbobster, 2019-04-23:
[getgnuradio.sh.txt](https://github.com/gnuradio/gnuradio/files/3109299/getgnuradio.sh.txt)

marcusmueller, 2024-01-22:
hey @maitbot; I know it's something that you're using and had fought before; do you have input on this?

closed 2024-02-14 by marcusmueller