gnuradio/gnuradio #2068: make test (qa_zeromq) fail

Red-Owl, 2018-10-05:
Start 356: qa_zeromq_pub
356/360 Test #356: qa_zeromq_pub ......................................***Failed    1.73 sec
        Start 357: qa_zeromq_pubsub
357/360 Test #357: qa_zeromq_pubsub ...................................***Failed    1.70 sec
        Start 358: qa_zeromq_pushpull
358/360 Test #358: qa_zeromq_pushpull .................................***Failed    1.63 sec
        Start 359: qa_zeromq_reqrep
359/360 Test #359: qa_zeromq_reqrep ...................................***Failed    1.67 sec
        Start 360: qa_zeromq_sub
360/360 Test #360: qa_zeromq_sub ......................................***Failed    1.64 sec

97% tests passed, 10 tests failed out of 360

Total Test time (real) = 582.72 sec

The following tests FAILED:
        242 - qa_polar_decoder_sc (Failed)
        243 - qa_polar_decoder_sc_list (Failed)
        244 - qa_polar_decoder_sc_systematic (Failed)
        245 - qa_polar_encoder (Failed)
        246 - qa_polar_encoder_systematic (Failed)
        356 - qa_zeromq_pub (Failed)
        357 - qa_zeromq_pubsub (Failed)
        358 - qa_zeromq_pushpull (Failed)
        359 - qa_zeromq_reqrep (Failed)
        360 - qa_zeromq_sub (Failed)
Errors while running CTest
Makefile:105: set di istruzioni per l'obiettivo "test" non riuscito

michaelld, 2018-10-05:
What OS? What version of GNU Radio (git commit or release version)?

michaelld, 2018-10-08:
VOLK: Which version of Volk (release, GIT commit)? I'd use the current GIT master if I were you ... it contains fixes for newer GCC & some numerical issues with some of the kernels. What does the following return: `ctest -V -R qa_volk_32f_log2_32f`? Ditto for the others?

GR: what does the following return: `ctest -V -R qa_polar_decoder_sc`?

GR: zeromq success depends partly on how networking is configured; on my various test computers, zeromq testing can be complete failure, mostly failure, mostly success, or complete success. All depends on the OS & networking settings, in my testing.

Red-Owl, 2018-10-11:
i have redownload gnuradio:  --recursive https://github.com/gnuradio/gnuradio.git
 and not error make test gnuradio and volk

michaelld, 2018-10-12:
OK ... so what i think you're saying is that it's working OK & we can close this issue, yes?

michaelld, 2018-10-17:
No response for a few days, so I'd say to close this issue (I can't).

btashton, 2018-11-15:
Can we close this out.  The recent changes to zmq tests seem to make them a lot more reliable and this seems to not be reproducible now.

closed 2018-11-15 by noc0lour

pianoman730, 2019-05-23:
@noc0lour I just had this same issue...
```
97% tests passed, 10 tests failed out of 359

Total Test time (real) = 188.84 sec

The following tests FAILED:
	242 - qa_polar_decoder_sc (Failed)
	243 - qa_polar_decoder_sc_list (Failed)
	244 - qa_polar_decoder_sc_systematic (Failed)
	245 - qa_polar_encoder (Failed)
	246 - qa_polar_encoder_systematic (Failed)
	355 - qa_zeromq_pub (Failed)
	356 - qa_zeromq_pubsub (Failed)
	357 - qa_zeromq_pushpull (Failed)
	358 - qa_zeromq_reqrep (Failed)
	359 - qa_zeromq_sub (Failed)
Errors while running CTest
make: *** [test] Error 8
```

I'm running on centos 7. I'm wondering if these fails will inhibit my work completely. How to proceed?

michaelld, 2019-05-23:
What does `ctest -V -R qa_zeromq_pub` return?

pianoman730, 2019-05-23:
I ran `ctest -V -R qa_zeromq_pub` and `ctest -V -R qa_polar_decoder_sc` and found I needed to `pip install scipy zmq` and they passed. Thanks!

michaelld, 2019-05-23:
YW! Glad that helped you figure out the issue & thanks for reporting back! Leaving this issue as closed.

sam1ee, 2019-06-10:
> I ran `ctest -V -R qa_zeromq_pub` and `ctest -V -R qa_polar_decoder_sc` and found I needed to `pip install scipy zmq` and they passed. Thanks!

I ran the same pip you did, thanks. This should be sticky and pushed to the top of this p