gnuradio/gnuradio #2075: OOTs don't work with 3.8

mbr0wn, 2018-10-11:
I tried to create an OOT with current master branch, at it fails. Two reproduce, create an OOT with modtool and add a block. Then, the following issues will pop up:

1. Finding GNU Radio will fail:

```
CMake Error at CMakeLists.txt:144 (find_package):
  Could not find a configuration file for package "Gnuradio" that is
  compatible with requested version "3.7.2".

  The following configuration files were considered but not accepted:

    /home/mbr0wn/prefix/default/lib/cmake/gnuradio/GnuradioConfig.cmake, version: 3.8.git.0
```

This might simply be because '3.8.git.0' is not considered a comparable version string. However, if I change the top-level CMakeLists.txt in the OOT like this, it passes:

```cmake
find_package(Gnuradio "3.8" REQUIRED)
```

2. Doxygen fails

```
Traceback (most recent call last):                                                                                       [9/202]
  File "/home/mbr0wn/tmp/gr-foo/docs/doxygen/swig_doc.py", line 330, in <module>
    make_swig_interface_file(di, swigdocfilename, custom_output=custom_output)
  File "/home/mbr0wn/tmp/gr-foo/docs/doxygen/swig_doc.py", line 253, in make_swig_interface_file
    blocks = di.in_category(Block)
  File "/home/mbr0wn/tmp/gr-foo/docs/doxygen/doxyxml/base.py", line 143, in in_category
    self.confirm_no_error()
  File "/home/mbr0wn/tmp/gr-foo/docs/doxygen/doxyxml/base.py", line 209, in confirm_no_error
    self.check_parsed()
  File "/home/mbr0wn/tmp/gr-foo/docs/doxygen/doxyxml/base.py", line 206, in check_parsed
    self._parse()                                                                                                    
  File "/home/mbr0wn/tmp/gr-foo/docs/doxygen/doxyxml/doxyindex.py", line 57, in _parse
    self._members += converted.members()
  File "/home/mbr0wn/tmp/gr-foo/docs/doxygen/doxyxml/base.py", line 177, in members
    self.confirm_no_error()
  File "/home/mbr0wn/tmp/gr-foo/docs/doxygen/doxyxml/base.py", line 209, in confirm_no_error
    self.check_parsed()
  File "/home/mbr0wn/tmp/gr-foo/docs/doxygen/doxyxml/base.py", line 206, in check_parsed
    self._parse()
  File "/home/mbr0wn/tmp/gr-foo/docs/doxygen/doxyxml/doxyindex.py", line 242, in _parse
    self.set_descriptions(self._retrieved_data.compounddef)
  File "/home/mbr0wn/tmp/gr-foo/docs/doxygen/doxyxml/doxyindex.py", line 84, in set_descriptions
    bd = description(getattr(parse_data, 'briefdescription', None))
  File "/home/mbr0wn/tmp/gr-foo/docs/doxygen/doxyxml/text.py", line 40, in description
    return description_bit(obj).strip()
  File "/home/mbr0wn/tmp/gr-foo/docs/doxygen/doxyxml/text.py", line 47, in description_bit
    contents = [description_bit(item) for item in obj.content_]
  File "/home/mbr0wn/tmp/gr-foo/docs/doxygen/doxyxml/text.py", line 50, in description_bit
    result = description_bit(obj.value)
  File "/home/mbr0wn/tmp/gr-foo/docs/doxygen/doxyxml/text.py", line 54, in description_bit
    raise Exception('Expecting a string or something with content, content_ or value attribute')
Exception: Expecting a string or something with content, content_ or value attribute
swig/CMakeFiles/foo_swig_swig_doc.dir/build.make:60: recipe for target 'swig/foo_swig_doc.i' failed
make[2]: *** [swig/foo_swig_doc.i] Error 1
CMakeFiles/Makefile2:238: recipe for target 'swig/CMakeFiles/foo_swig_swig_doc.dir/all' failed
make[1]: *** [swig/CMakeFiles/foo_swig_swig_doc.dir/all] Error 2
```

This can be bypassed by disabling Doxygen in CMake (`-DENABLE_DOXYGEN=OFF`), although that's not a fix.

3. Something something SWIG

Next, run ctest:

```
2: Test command: /bin/sh "/home/mbr0wn/tmp/gr-foo/build/python/qa_square_ff_test.sh"
2: Test timeout computed to be: 9.99988e+06
2: Traceback (most recent call last):
2:   File "/home/mbr0wn/tmp/gr-foo/python/qa_square_ff.py", line 23, in <module>
2:     from gnuradio import gr, gr_unittest
2:   File "/home/mbr0wn/prefix/default/lib/python3.6/dist-packages/gnuradio/gr/__init__.py", line 39, in <module>
2:     from .runtime_swig import *
2:   File "/home/mbr0wn/prefix/default/lib/python3.6/dist-packages/gnuradio/gr/runtime_swig.py", line 117
2:     def value(self) -> "PyObject *":
2:                     ^
2: SyntaxError: invalid syntax
2/2 Test #2: qa_square_ff .....................***Failed    0.02 sec
```

mbr0wn, 2018-10-11:
The latter problem is because I was using GNU Radio with Py3k (which is the default if available), but the Python executable for OOTs gets set to Python 2.

primercuervo, 2018-11-13:
being this labeled as high, I wonder if getting one's hands dirty here might be done after getting done with https://github.com/gnuradio/gnuradio/pull/1873 (which I'd really like to see merged, ahem, @marcusmueller )

noc0lour, 2018-11-13:
One problem inherent Problem with GR in that respect is the lack of version information about which python the installed GR has used. Maybe that's the legit reason to switch to modern CMake's config information.

hcab14, 2018-11-18:
This is great work. As I am woking on some OOT modules, the broken OOT module functionality is the only thing which keeps me (and maybe also others) from using the master branch of gnuradio.

jdemel, 2019-01-01:
I ran into the same problem the other day.

`gr_modtool` does currently create outdated configuration. It would be very helpful to make modtool and GR configuration more concurrent. I'd suggest to try to move GR CMake magic etc. which is currently mostly in the appropriate `CMakeLists.txt` files into cmake modules which are installed.
I see 2 potential benefits. We could reduce code duplication. OOT and GR configuration would be more  in sync.

One thing I really like about the current `CMakeLists.txt` is that it clearly states minimum versions for most (all?) dependencies.

Actually, I started to go through modtool and tried to fix some issues in order to be able to create an up-to-date OOT, but then I came across #1873 and decided to wait until this is merged. All in all I see this issue as a post- #1873-merge task.

noc0lour, 2019-03-08:
@mbr0wn can you verify with the lasted developments on master?

drmpeg, 2019-03-10:
gr_modtool almost works as of commit https://github.com/gnuradio/gnuradio/commit/bdaee5982f40f739dcb58a4963a813fb2746cf49. The only remaining problems (that I know of) are detecting Python versions and building for Python2.

Detecting Python versions can be fixed by adding these lines to gr-utils/python/modtool/templates/gr-newmod/CMakelists.txt
```
set(GR_PYTHON_MIN_VERSION "2.7")
set(GR_PYTHON3_MIN_VERSION "3.4")
```
However, the build fails for Python 2 when generating Doxygen. Therefore, I didn't include the above fix in my PR because I'm guessing there's a better solution.

noc0lour, 2019-03-10:
The python problem should be solved by setting `PYTHON_EXECUTABLE` in the gnuradio config cmake. This will hardcode the python interpreter used by the main GNU Radio installation. I'll checkout the python2 issue.

drmpeg, 2019-03-10:
Just to be clear, you mean ```PYTHON_EXECUTABLE``` should be set in the<install_dir>/lib/cmake/gnuradio/GnuradioConfig.cmake file?

noc0lour, 2019-03-10:
Yes, that's what is done. Since letting depending libraries select PYTHON_EXECUTABLE doesn't make sense. Maybe we should still leave an override switch and not hardcode it...

jdemel, 2019-03-10:
@drmpeg I'd argue that there are 3 outstanding issues.
1. `PYTHON_EXECUTABLE` should be set in `GnuradioConfig.cmake`.
2. Thrift support for OOTs should be fixed with #2393 by @marcusmueller.
3. Python2 build. Though, I'd argue that this is the least problematic one for most users.

Also, I try to figure out how to set `PYTHON_EXECUTABLE` in `GnuradioConfig.cmake` during build. @noc0lour how is this supposed to work?

noc0lour, 2019-03-10:
#2393 has everything in it.

noc0lour, 2019-03-10:
The `ENABLE_GRC` setting still has a bug though.

closed 2019-03-24 by noc0lour