gnuradio/gnuradio #2078: Not Stop Flowchart

Red-Owl, 2018-10-11:
Gnuradio 3.8tech-preview-8-g30c9ad08 (Python 3.6.7rc1)
not stop subprocess and not replay.
Close gnuradio and reopen :(

gruradio from console:

root@meo:/home/gufo/Scrivania/Nuova cartella/gr-fosphor/build# gnuradio-companion
/usr/local/lib/python3.6/dist-packages/gnuradio/grc/core/FlowGraph.py:21: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
  import imp

(gnuradio-companion:3171): Gtk-WARNING **: 18:55:35.779: Theme parsing error: gtk-widgets.css:153:27: The style property GtkButton:image-spacing is deprecated and shouldn't be used anymore. It will be removed in a future version

(gnuradio-companion:3171): Gtk-WARNING **: 18:55:35.779: Theme parsing error: gtk-widgets.css:154:26: The style property GtkButton:inner-border is deprecated and shouldn't be used anymore. It will be removed in a future version

(gnuradio-companion:3171): Gtk-WARNING **: 18:55:35.779: Theme parsing error: gtk-widgets.css:578:27: The style property GtkButton:image-spacing is deprecated and shouldn't be used anymore. It will be removed in a future version

(gnuradio-companion:3171): Gtk-WARNING **: 18:55:35.779: Theme parsing error: gtk-widgets.css:579:28: The style property GtkWidget:interior-focus is deprecated and shouldn't be used anymore. It will be removed in a future version

(gnuradio-companion:3171): Gtk-WARNING **: 18:55:35.779: Theme parsing error: gtk-widgets.css:580:28: The style property GtkButton:default-border is deprecated and shouldn't be used anymore. It will be removed in a future version

(gnuradio-companion:3171): Gtk-WARNING **: 18:55:35.779: Theme parsing error: gtk-widgets.css:581:26: The style property GtkButton:inner-border is deprecated and shouldn't be used anymore. It will be removed in a future version

(gnuradio-companion:3171): Gtk-WARNING **: 18:55:35.781: Theme parsing error: gtk-widgets.css:1357:25: The style property GtkRange:slider-width is deprecated and shouldn't be used anymore. It will be removed in a future version

(gnuradio-companion:3171): Gtk-WARNING **: 18:55:35.781: Theme parsing error: gtk-widgets.css:1358:25: The style property GtkRange:stepper-size is deprecated and shouldn't be used anymore. It will be removed in a future version

(gnuradio-companion:3171): Gtk-WARNING **: 18:55:35.781: Theme parsing error: gtk-widgets.css:1359:34: The style property GtkScrollbar:min-slider-length is deprecated and shouldn't be used anymore. It will be removed in a future version

(gnuradio-companion:3171): Gtk-WARNING **: 18:55:35.781: Theme parsing error: gtk-widgets.css:1360:28: The style property GtkRange:stepper-spacing is deprecated and shouldn't be used anymore. It will be removed in a future version

(gnuradio-companion:3171): Gtk-WARNING **: 18:55:35.781: Theme parsing error: gtk-widgets.css:1361:34: The style property GtkRange:trough-under-steppers is deprecated and shouldn't be used anymore. It will be removed in a future version
DEBUG:grc:Loading platform
WARNING:gnuradio.grc.core.platform:variable_struct.block.yml                block: warn: Ignoring extra key 'var_value'
DEBUG:grc:Loading application
DEBUG:grc:Running

(gnuradio-companion:3171): GLib-GIO-CRITICAL **: 18:55:36.073: g_dbus_proxy_new_sync: assertion 'G_IS_DBUS_CONNECTION (connection)' failed

(gnuradio-companion:3171): Gtk-WARNING **: 18:55:36.134: actionhelper: action app.flowgraph.open_recent can't be activated due to parameter type mismatch (parameter type s, target type NULL)

(gnuradio-companion:3171): IBUS-WARNING **: 18:55:36.247: The owner of /home/gufo/.configtde/ibus/bus is not root!
<<< Welcome to GNU Radio Companion 3.8tech-preview-8-g30c9ad08 >>>

Block paths:
        /usr/local/share/gnuradio/grc/blocks

Loading: "/home/gufo/Scrivania/untitled.grc"
>>> Done
/usr/lib/python3.6/subprocess.py:766: ResourceWarning: subprocess 3182 is still running
  ResourceWarning, source=self)

Generating: '/home/gufo/Scrivania/top_block.py'

Executing: /usr/bin/python3 -u /home/gufo/Scrivania/top_block.py
/usr/lib/python3.6/subprocess.py:596: ResourceWarning: unclosed file <_io.BufferedWriter name=12>
  _cleanup()
/usr/lib/python3.6/subprocess.py:596: ResourceWarning: unclosed file <_io.BufferedReader name=13>
  _cleanup()
/usr/lib/python3.6/subprocess.py:596: ResourceWarning: unclosed file <_io.BufferedReader name=15>
  _cleanup()

Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3.6/dist-packages/gnuradio/grc/gui/Executor.py", line 86, in run
    r = self.process.stdout.read(1024).decode('utf-8','ignore')
AttributeError: 'str' object has no attribute 'decode'

Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/gnuradio/grc/gui/Application.py", line 719, in _handle_action
    page.term_proc()
AttributeError: 'Page' object has no attribute 'term_proc'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/gnuradio/grc/gui/Application.py", line 721, in _handle_action
    print("could not terminate process: %d" % page.get_proc().pid)
AttributeError: 'Page' object has no attribute 'get_proc'
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/gnuradio/grc/gui/Application.py", line 719, in _handle_action
    page.term_proc()
AttributeError: 'Page' object has no attribute 'term_proc'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/gnuradio/grc/gui/Application.py", line 721, in _handle_action
    print("could not terminate process: %d" % page.get_proc().pid)
AttributeError: 'Page' object has no attribute 'get_proc'
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/gnuradio/grc/gui/Application.py", line 719, in _handle_action
    page.term_proc()
AttributeError: 'Page' object has no attribute 'term_proc'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/gnuradio/grc/gui/Application.py", line 721, in _handle_action
    print("could not terminate process: %d" % page.get_proc().pid)
AttributeError: 'Page' object has no attribute 'get_proc'
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/gnuradio/grc/gui/Application.py", line 719, in _handle_action
    page.term_proc()
AttributeError: 'Page' object has no attribute 'term_proc'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/gnuradio/grc/gui/Application.py", line 721, in _handle_action
    print("could not terminate process: %d" % page.get_proc().pid)
AttributeError: 'Page' object has no attribute 'get_proc'

(gnuradio-companion:3171): Gtk-CRITICAL **: 18:55:42.867: gtk_main_quit: assertion 'main_loops != NULL' failed
root@meo:/home/gufo/Scrivania/Nuova cartella/gr-fosphor/build#

michaelld, 2018-10-12:
Thank you for the issue report. As I'm sure you're aware, you're using the first preview of the 3.8 release, which is chock-full-o-bugs. We're working hard to fix these issues; this one looks like it a has a pull request that should fix it, #2079 ... if you're able to test this PR, please do so & report back. If not, then keep updating you GR GIT repo & doing more testing, and let us know when issues are fixed, and also open new issues as you encounter new issues. Thanks for your understanding & patience as we work out the kinks in GR before the 3.8 release!

dl1ksv, 2018-10-13:
This is a long known issue and there exist at least 3 or 4 pr's to fix this issue.

For instance see https://github.com/gnuradio/gnuradio/pull/2024

michaelld, 2018-10-17:
No response for a few days & this is a known issue with possible fixes already in the queue ... I'd say to close this issue.

777arc, 2018-10-27:
Agreed, closing.

PR #2079 (2018-11-12): GRC: fix failing thread run
In Py3k a string decode might trigger an error because it is already a unicode string. This is not the case in Py2k land. Thus, add a check.

This should fix #2078 .
closed 2018-10-27 by 777arc