gnuradio/gnuradio #2083: Request json to file .GRC

Red-Owl, 2018-10-12:
I have read a new file .grc if is possible to use a json file, this is readable for human and machine and is compatible to database server (mysql postgree php...) and other application and have all type of obj int string etc...
TNX

777arc, 2018-10-17:
.grc files use to be xml, and now they are yaml (using an auto-converter to keep backwards compatibility), so the only way to support json files would be if you created a json-to-yaml grc converter that supported all of grc's functionality.

noc0lour, 2018-10-17:
`yaml` is a superset of `json`. So, if we are not using special `yaml` syntax which is not available in `json` you can just convert `yaml` to `json` and store it in a database and export it back to `yaml` to import it into GNU Radio.

mbr0wn, 2018-12-13:
I think the converter is a good suggestion. Closing this.

closed 2018-12-13 by mbr0wn