gnuradio/gnuradio #2086: GNURadio/master not building for 32 bit machines

balister, 2018-10-13:
Confirmed on 32 bit ARM and 32 bit x86. This is with a patch from Michael Dickens.

| -- Check size of unsigned long                                       [37/1900]
| -- Check size of unsigned long - done                                         
| -- Check size of uint64_t                                                     
| -- Check size of uint64_t - done                                              
| -- FindMPLIB: sizeof(unsigned long) [4] is not equal to sizeof (uint64_t) [8] 
and must be for the MPLIB (GMP or MPIR) to build.                               | CMake Error at cmake/Modules/FindMPLIB.cmake:35 (message):                    
|   FindMPLIB: cannot find a type conversion for MPLIB (GMP or MPIR).           | Call Stack (most recent call first):                                          
|   CMakeLists.txt:480 (find_package)    

Without patch:

/home/balister/opensdr/ti-thud/build/tmp-glibc/work/i586-oe-linux/gnuradio/3.8.0
+gitAUTOINC+007923c64d-r0/git/gnuradio-runtime/lib/block_executor.cc: In functio
n 'bool gr::propagate_tags(gr::block::tag_propagation_policy_t, gr::block_detail
*, const std::vector<long long unsigned int>&, double, mpq_class&, bool, std::ve
ctor<gr::tag_t>&, long int)':
/home/balister/opensdr/ti-thud/build/tmp-glibc/work/i586-oe-linux/gnuradio/3.8.0
+gitAUTOINC+007923c64d-r0/git/gnuradio-runtime/lib/block_executor.cc:149:37: err
or: ambiguous overload for 'operator*' (operand types are 'uint64_t' {aka 'long 
long unsigned int'} and 'mpq_class' {aka '__gmp_expr<__mpq_struct [1], __mpq_str
uct [1]>'})
             offset = new_tag.offset * mp_rrate + one_half;
                      ~~~~~~~~~~~~~~~^~~~~~~~~~

michaelld, 2018-10-13:
FYI that the patch type casts certain GMP / MPIR function call arguments from `uint64_t`(typically `unsigned long long`) to `unsigned long`, as this latter is the max type size allowed by GMP / MPIR. On modern compilers, these are different types, and the compiler won't do implicit type casting on them, even if they are the same bit-size. On (at least some) 32-bit systems, things get even worse since `unsigned long` is just 32 bits wide (while clearly `uint64_t` is 64 bits wide). For better or worse, what I might do is change the patch's error to a warning since I believe that change will allow GR to build -- but I have no idea if/how well GR will execute.

michaelld, 2018-11-15:
I believe we can close this now, yes?

noc0lour, 2018-11-15:
Yep. Thanks for reporting and helping figuring this out!

closed 2018-11-15 by noc0lour