gnuradio/gnuradio #2087: SoapySDR HackRF driver receiver issue

markmower, 2018-10-15:
Hi

I am using Hackrf one and the SoapySDR driver for Hackrf one. when using a flow graph with transmit everything works fine. For a receiver flow graph I get the following error self.osmosdr_source_0.set_iq_balance_mode(2, 0)
  File "/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/osmosdr/osmosdr_swig.py", line 1779, in set_iq_balance_mode
    return _osmosdr_swig.source_sptr_set_iq_balance_mode(self, mode, chan)
RuntimeError: soapy_source_c::set_iq_balance_mode() not supported

using Hackrf on its own it works fine. I want to use SoapySDR driver for Hackrf as it allows me to have one flow graph with rx and tx.

could you please advise on how to solve the above issue

michaelld, 2018-10-15:
The short answer is do not use IQ balance with SoapySDR until it is enabled. See also this code: https://github.com/osmocom/gr-osmosdr/blob/master/lib/soapy/soapy_source_c.cc#L310 , which explicitly disables IQ balance in the gr-osmosdr interface to SoapySDR. If you believe you can fix the issue, please do so & then create a pull request for the effected projects.

NOTE: Your query is a little off-topic, as it pertains to HacKRF, SoapySDR, and gr-osmosdr, and how they interact -- not GNU Radio per se. You would be better off querying their developers / lists, and, really, I'd recommend you do that anyway since the issue is one of updating some or all of these projects to support IQ balance.

There is nothing we (GR developers) can do here, as none of these projects are under our purview; please close this issue and move the discussion to one of those projects' issue tracker.

markmower, 2018-10-15:
Hi

Ok understood, thanks for reply and and specifically IQ balance, apologies for placing in wrong project I was unsure where the problem is and have been trying to figure out.

Again thanks for the reply it has been very helpful and I will close the issue.

Kind regards

Mark

closed 2018-10-15 by markmower