gnuradio/gnuradio #2089: gnuradio-3.7.13.4 build issue with thrift version 0.11.0

wrthissell, 2018-10-17:
Dear Colleagues,
      Thank you in advance for helping me resolve the following issue.  
      I am trying to build the above listed version of gnuradio in a Centos 7.5 machine using the conda version of python because I have build issues with all of the rebuilt dependencies with the system python.  I had serious issues getting a thrift rpm built with version >= 0.9.2, but I finally got the attached thrift version 0.11.0 built and installed.  I was unable to get the rawhide src rpm for thrift 0.10 to build correctly on Centos 7.5 without error.  I also installed a conda-forge version of thrift 0.11 to enable the build of gnuradio because of errors with the system python with pygtk with the available pygtk2 rpm.  The attached tar files include the thrift src and rpm builds, as well as build error related to thrift 0.11.0, which is summarized below:

[wrthissell@LAPTOP-BKIJEPGK ~]$ uname -r
4.18.10-1.el7.elrepo.x86_64

rpmbuid --ba gnuradio.spec
...
[20181016 gnuradio rpmbuild issue.tar.gz](https://github.com/gnuradio/gnuradio/files/2485391/20181016.gnuradio.rpmbuild.issue.tar.gz)

config-info.dir/link.txt --verbose=1
/usr/bin/c++   -O2 -g -pipe -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches   -m64 -mtune=generic  -fvisibility=hidden -Wsign-compare -Wno-uninitialized -std=c++98 -DNDEBUG   -Wl,-z,relro  CMakeFiles/gnuradio-config-info.dir/gnuradio-config-info.cc.o  -o gnuradio-config-info -rdynamic ../lib/libgnuradio-runtime-3.7.13.4.so.0.0.0 -lboost_date_time-mt -lboost_program_options-mt -lboost_filesystem-mt -lboost_system-mt -lboost_regex-mt -lboost_thread-mt -lthrift ../lib/pmt/libgnuradio-pmt-3.7.13.4.so.0.0.0 -lboost_date_time-mt -lboost_program_options-mt -lboost_filesystem-mt -lboost_system-mt -lboost_regex-mt -lboost_thread-mt ../../volk/lib/libvolk.so.1.4 -ldl -lorc-0.4 -llog4cpp -lrt -Wl,-rpath,/home/wrthissell/rpmbuild/BUILD/gnuradio-3.7.13.4/build/gnuradio-runtime/lib:/home/wrthissell/rpmbuild/BUILD/gnuradio-3.7.13.4/build/gnuradio-runtime/lib/pmt:/home/wrthissell/rpmbuild/BUILD/gnuradio-3.7.13.4/build/volk/lib: 
../lib/libgnuradio-runtime-3.7.13.4.so.0.0.0: undefined reference to `apache::thrift::server::TSimpleServer::TSimpleServer(boost::shared_ptr<apache::thrift::TProcessor> const&, boost::shared_ptr<apache::thrift::transport::TServerTransport> const&, boost::shared_ptr<apache::thrift::transport::TTransportFactory> const&, boost::shared_ptr<apache::thrift::protocol::TProtocolFactory> const&)'
../lib/libgnuradio-runtime-3.7.13.4.so.0.0.0: undefined reference to `apache::thrift::server::TThreadPoolServer::TThreadPoolServer(boost::shared_ptr<apache::thrift::TProcessor> const&, boost::shared_ptr<apache::thrift::transport::TServerTransport> const&, boost::shared_ptr<apache::thrift::transport::TTransportFactory> const&, boost::shared_ptr<apache::thrift::protocol::TProtocolFactory> const&, boost::shared_ptr<apache::thrift::concurrency::ThreadManager> const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [gnuradio-runtime/apps/gnuradio-config-info] Error 1
make[2]: Leaving directory `/home/wrthissell/rpmbuild/BUILD/gnuradio-3.7.13.4/build'
make[1]: *** [gnuradio-runtime/apps/CMakeFiles/gnuradio-config-info.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
Linking CXX shared module _pmt_swig.so
cd /home/wrthissell/rpmbuild/BUILD/gnuradio-3.7.13.4/build/gnuradio-runtime/swig && /usr/bin/cmake -E cmake_link_script CMakeFiles/_pmt_swig.dir/link.txt --verbose=1
/usr/bin/c++  -fPIC -O2 -g -pipe -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches   -m64 -mtune=generic  -fvisibility=hidden -Wsign-compare -Wno-uninitialized -std=c++98 -DNDEBUG  -Wl,-z,relro  -shared -Wl,-soname,_pmt_swig.so -o _pmt_swig.so CMakeFiles/_pmt_swig.dir/pmt_swigPYTHON_wrap.cxx.o -lpython2.7 ../lib/pmt/libgnuradio-pmt-3.7.13.4.so.0.0.0 ../lib/libgnuradio-runtime-3.7.13.4.so.0.0.0 -llog4cpp ../lib/pmt/libgnuradio-pmt-3.7.13.4.so.0.0.0 -lthrift -lboost_date_time-mt -lboost_program_options-mt -lboost_filesystem-mt -lboost_system-mt -lboost_regex-mt -lboost_thread-mt -llog4cpp ../../volk/lib/libvolk.so.1.4 -ldl -lorc-0.4 -lrt -Wl,-rpath,/home/wrthissell/rpmbuild/BUILD/gnuradio-3.7.13.4/build/gnuradio-runtime/lib/pmt:/home/wrthissell/rpmbuild/BUILD/gnuradio-3.7.13.4/build/gnuradio-runtime/lib:/home/wrthissell/rpmbuild/BUILD/gnuradio-3.7.13.4/build/volk/lib: 
make[2]: Leaving directory `/home/wrthissell/rpmbuild/BUILD/gnuradio-3.7.13.4/build'
/usr/bin/cmake -E cmake_progress_report /home/wrthissell/rpmbuild/BUILD/gnuradio-3.7.13.4/build/CMakeFiles 
[ 24%] Built target _pmt_swig
make[1]: Leaving directory `/home/wrthissell/rpmbuild/BUILD/gnuradio-3.7.13.4/build'
make: *** [all] Error 2
error: Bad exit status from /var/tmp/rpm-tmp.m2Bxbf (%build)

RPM build errors:
    Bad exit status from /var/tmp/rpm-tmp.m2Bxbf (%build)
(pygtk) [wrthissell@LAPTOP-BKIJEPGK SPECS]$ 

     I suspect that a patch is required in gnuradio to make it build with thrift 0.11.  I recommend that you consider making such a patch.
     Thank you again for helping me resolve this issue.

michaelld, 2018-10-17:
Thank you for the issue report. I wouldn't doubt if fixes in GR are required for the latest Thrift release. Hopefully you can do without Thrift in the meantime until a fix is found ;)

tmancill, 2019-01-01:
Hi.  The problem compiling against Thrift 0.11.0 also affects Debian (Debian bug [#890339](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=890339)).

yakkonaut, 2020-06-16:
FYI this is still outstanding. I was able to build current master (3.9.0.0) without thrift enabled- and I don't really use control port so I'm not sure how much that impacts performance, but will confirm there seem to be numerous calls to deprecated/out-of-date function prototypes when attempting to build against thrift 0.10.0.

yakkonaut, 2020-06-16:
https://github.com/gnuradio/gnuradio/issues/3515

marcusmueller, 2023-12-22:
We require Thrift v 0.13.0+ by now, so this can be closed

closed 2023-12-22 by marcusmueller