gnuradio/gnuradio #2097: grc gets a lot of DeprecationWarning

dl1ksv, 2018-10-19:
Starting grc produces many DeprecationWarning: invalid escape sequence \s processing filter_swig.py.

Here are only a few of them:

/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/filter/filter_swig.py:1480: DeprecationWarning: invalid escape sequence \s
  """
/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/filter/filter_swig.py:1517: DeprecationWarning: invalid escape sequence \s
  """
/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/filter/filter_swig.py:1560: DeprecationWarning: invalid escape sequence \s
  """
/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/filter/filter_swig.py:1590: DeprecationWarning: invalid escape sequence \s
  """
/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/filter/filter_swig.py:1627: DeprecationWarning: invalid escape sequence \s
  """
/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/filter/filter_swig.py:1670: DeprecationWarning: invalid escape sequence \s
  """
/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/filter/filter_swig.py:1700: DeprecationWarning: invalid escape sequence \s
  """

btashton, 2018-10-29:
I'm not quite sure what we want to do here.  We technically are not doing the right thing for python2 or 3
The issue is the filter package has latex in the docstrings.  Swig4 has some changes that might generate raw strings in these cases, but it is not clear to me if that would really help and most people are going to be on swig3

Some projects are doing the escaping via cmake, which I'm also not a fan of
https://github.com/openturns/openturns/pull/786
 
and example is this:
```
     * The input and output satisfy a difference equation of the form
     \f[
       y[n] - \sum_{k=1}^{M} a_k y[n-k] = \sum_{k=0}^{N} b_k x[n-k]
     \f]

     \xmlonly
     y[n] - \sum_{k=1}^{M} a_k y[n-k] = \sum_{k=0}^{N} b_k x[n-k]
     \endxmlonly

     * with the corresponding rational system function
```
generating this
```
The input and output satisfy a difference equation of the form 

 
y[n] - \sum_{k=1}^{M} a_k y[n-k] = \sum_{k=0}^{N} b_k x[n-k]

with the corresponding rational system function 
```

btashton, 2018-10-30:
Looks like there is already a PR that should fix this #2047

dl1ksv, 2018-10-30:
@btashton No, this pr fixes a lot of depreciation warnigs, but not these above .

By the  way, what is it good for to use latex notation at this point ? Why not to use pure utf8 code ?

btashton, 2018-10-30:
You are not going to be able to represent these all with utf8, many documentation generators will properly parse latex and generate the images

dl1ksv, 2018-10-30:
Sure, but we are talking about the use in gnuradio and python. 
Do you think that

\sum_{k=1}^{M} a_k y[n-k] = \sum_{k=0}^{N} b_k x[n-k] 
is more readable than
sum( a(k)*y(n-k) ) , k=1,..,M ?

noc0lour, 2018-10-30:
The idea is to have formulas etc rendered by doxygen to be able to look at them in documentation: https://www.gnuradio.org/doc/doxygen/classgr_1_1filter_1_1iir__filter__ccc.html#details

noc0lour, 2018-10-30:
By default swig is taking the C++ docs and pulling them into docstrings (I think) which does not help in this case.

btashton, 2018-10-30:
@noc0lour I agree about keeping them in LaTex.  It would be nice if we could keep them in there, things like ipython will render them as well as if we ever use sphinx to generate documentation (I would be a huge fan...).

Right now the only thing that is escaped are double quotes and the latex markup.  If we dropped double quotes for single quotes we could just escape all the `\`  Several other projects have made it a rule to not include double quotes in docstrings and I think the benefit here outweighs the cost.  Thoughts?

noc0lour, 2018-10-30:
We have a sphinx generator and docs generated by it are here: https://www.gnuradio.org/doc/sphinx/

btashton, 2018-10-30:
@noc0lour The PR I just submitted should resolve this without having to put any restrictions in.