gnuradio/gnuradio #2100: Compile error on macOS 10.12.6

umi-neko, 2018-10-21:
Hello all,

I try to compile 3.8 preview on macOS, when compile the gnuradio-runtime package, there is some conversion errors.
Here is the error message:
[build.log](https://github.com/gnuradio/gnuradio/files/2499268/build.log)
And the source I used is download from releases page 
https://github.com/gnuradio/gnuradio/archive/3.8tech-preview.tar.gz

michaelld, 2018-10-21:
This is no specifically an OSX issue ... for better or worse! There is a possible patch to fix this issue on both 64-bit and 32-bit OSs with the issue. See also issue https://github.com/gnuradio/gnuradio/issues/2086 and https://github.com/noc0lour/gnuradio/tree/fix_32bit_uint64t_gmp .

michaelld, 2018-10-21:
The latter has 2 commits that should fix this issue for both really-recent 64-bit as well as pretty much any 32-bit. We're still testing / verify this fix; your testing would be welcome if you're able to do so.

umi-neko, 2018-10-22:
Thank you very much, I will try it tonight.

umi-neko, 2018-10-25:
I refresh the code, and the compile error gone. Maybe I can close this issue now. 
But there is some new compile error about thrift, maybe version not correct? 
If needed I will open a new issue.

closed 2018-10-25 by umi-neko