gnuradio/gnuradio #2107: cmake errors (master branch)

dmj2144, 2018-10-24:
Please bear with me as I am a total gnuradio newb.  Recently, I clone'd the master branch and attempted to compile/install on my CentOS 7 box.  Everything went well !  Then, I followed this link (https://wiki.gnuradio.org/index.php/OutOfTreeModules) to create and out-of-tree-module.  At this point, I went to "mkdir build && cd build && cmake -DPYTHON_EXECUTABLE=/path/to/pyenv/executable ..".  I received these error messages

-- Found SWIG version 4.0.0.
Traceback (most recent call last):
  File "<string>", line 2, in <module>
TypeError: Unicode-objects must be encoded before hashing
CMake Warning (dev) at cmake/Modules/UseSWIG.cmake:199 (add_executable):
  Policy CMP0037 is not set: Target names should not be reserved and should
  match a validity pattern.  Run "cmake --help-policy CMP0037" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.

  The target name "/export/home/djansky/projects/gr-howto/build/swig/.cpp" is
  reserved or not valid for certain CMake features, such as generator
  expressions, and may result in undefined behavior.
Call Stack (most recent call first):
  cmake/Modules/UseSWIG.cmake:256 (SWIG_ADD_SOURCE_TO_MODULE)
  cmake/Modules/GrSwig.cmake:174 (SWIG_ADD_MODULE)
  swig/CMakeLists.txt:51 (GR_SWIG_MAKE)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Error at cmake/Modules/UseSWIG.cmake:202 (add_custom_command):
  add_custom_command Wrong syntax.  A TARGET or OUTPUT must be specified.
Call Stack (most recent call first):
  cmake/Modules/UseSWIG.cmake:256 (SWIG_ADD_SOURCE_TO_MODULE)
  cmake/Modules/GrSwig.cmake:174 (SWIG_ADD_MODULE)
  swig/CMakeLists.txt:51 (GR_SWIG_MAKE)

I am confused as to where I am going wrong.  Any help would be appreciated.

Installs: CentOS 7, Python 3.4.0, cmake 3.12.3

dmj2144, 2018-10-26:
swig -version

SWIG Version 4.0.0

Compiled with g++ [x86_64-unknown-linux-gnu]

Configured options: +pcre

Please see http://www.swig.org for reporting bugs and further information

michaelld, 2018-10-26:
Well that's just strange. Can you try "reverting" to the latest release: 3.0.12? I've been using it for a long time & it's very stable and usable. Since there is no official SWIG 4.0.0 release as of yet, it's a little difficult to verify compatibility or lack thereof.

dmj2144, 2018-10-26:
I could try to revert but, reviewing the errors above I feel this is more of a cmake issue 2.8.x vs 3.12.x

michaelld, 2018-10-26:
Is your GR OOT module somewhere that I can access it to test / verify / debug? It's going to be much easier to help using actual code ...

dmj2144, 2018-10-26:
Checking for module SWIG
-- Found SWIG version 3.0.12.
-- Found SWIG: /usr/bin/swig  
-- Found PythonLibs: /export/home/djansky/.pyenv/versions/3.4.0/lib/libpython3.4m.so (found suitable version "2.7.9", minimum required is "2") 
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of size_t
-- Check size of size_t - done
-- Check size of unsigned int
-- Check size of unsigned int - done
-- Performing Test HAVE_WNO_UNUSED_BUT_SET_VARIABLE
-- Performing Test HAVE_WNO_UNUSED_BUT_SET_VARIABLE - Success
Traceback (most recent call last):
  File "<string>", line 2, in <module>
TypeError: Unicode-objects must be encoded before hashing
CMake Warning (dev) at cmake/Modules/UseSWIG.cmake:199 (add_executable):
  Policy CMP0037 is not set: Target names should not be reserved and should
  match a validity pattern.  Run "cmake --help-policy CMP0037" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.

  The target name "/export/home/djansky/projects/gr-howto/build/swig/.cpp" is
  reserved or not valid for certain CMake features, such as generator
  expressions, and may result in undefined behavior.
Call Stack (most recent call first):
  cmake/Modules/UseSWIG.cmake:256 (SWIG_ADD_SOURCE_TO_MODULE)
  cmake/Modules/GrSwig.cmake:174 (SWIG_ADD_MODULE)
  swig/CMakeLists.txt:51 (GR_SWIG_MAKE)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Error at cmake/Modules/UseSWIG.cmake:202 (add_custom_command):
  add_custom_command Wrong syntax.  A TARGET or OUTPUT must be specified.
Call Stack (most recent call first):
  cmake/Modules/UseSWIG.cmake:256 (SWIG_ADD_SOURCE_TO_MODULE)
  cmake/Modules/GrSwig.cmake:174 (SWIG_ADD_MODULE)
  swig/CMakeLists.txt:51 (GR_SWIG_MAKE)

dmj2144, 2018-10-26:
Also, I am only following the tutorial from https://wiki.gnuradio.org/index.php/OutOfTreeModules.

michaelld, 2018-10-27:
So I take it you're at the stage of "Using CMake": https://wiki.gnuradio.org/index.php/OutOfTreeModules#Using_CMake ... and, what you list is the result of the "cmake .." command?  I've never seen this particular CMake issue before, and I've been doing this for 13 years including the transition from GNU Autotools to CMake. Yes, I'm in agreement that it's probably a CMake issue. I'm running CMake 3.13.0-rc2 & don't see this issue; really very strange. Maybe there's some CMake cruft from a prior install & thus try reinstalling CMake?

dmj2144, 2018-10-29:
I found the issue !  Inside the UseSWIG.cmake that is generated from creating an OOTM there is a piece of python code that assumes you are using python 2.x.  

# hack to work around CMake bug in add_custom_command with multiple OUTPUT files

  file(RELATIVE_PATH reldir ${CMAKE_BINARY_DIR} ${CMAKE_CURRENT_BINARY_DIR})
  execute_process(
    COMMAND ${PYTHON_EXECUTABLE} -c "import re, hashlib
unique = hashlib.md5('${reldir}${ARGN}').hexdigest()[:5]
print(re.sub('\\W', '_', '${name} ${reldir} ' + unique))"
    OUTPUT_VARIABLE _target OUTPUT_STRIP_TRAILING_WHITESPACE
  )

The argument passed to md5() must be

'${reldir}${ARGN}'.encode('utf-8')

Instead of 

'${reldir}${ARGN}'

closed 2018-10-29 by dmj2144

michaelld, 2018-10-29:
Good catch! I'm guessing this fix or something like it is in the current GIT master, since it works with Python 2.7 or 3+. Thanks for reporting back!