gnuradio/gnuradio #2113: Numpy not being found (but numpy is installed)

mayorbobster, 2018-10-27:


mayorbobster, 2018-10-27:
Hello.  I am installing gnuradio (from github sources).  So far, all components are enabled except for:
doxygen (documents), gr-uhd ( I don't have ettis equipment), gr-zeromq, and gnuradio-companion.  The one I care about is gnuradio-companion.  I get:
Dependency NUMPY_FOUND = FALSE
but I wrote a script:
#!/usr/bin/env python2
try:
        import numpy
        HAS_NUMPY = True
except ImportError:
        HAS_NUMPY = False
print 'Numperical Python found: ' + str(HAS_NUMPY)
print 'Python version found: ' + str(numpy.__version__)
... and it gives me:
Numperical Python found: True
Python version found: 1.15.3
... I can install another version of numpy if that is required (I've tried a few already), but I keep getting:
Dependency NUMPY_FOUND = FALSE
My version of python is:
$ python -V
Python 2.7.15rc1
My system is Linux mint 19, I'm using cmake 
$ cmake -version
cmake version 3.12.3
.. I had to install a new version of cmake because I had to install a new version of curl for another application, cmake is dependent on curl, (and breaks if you don't rebuild it).  I've built other applications with this version of cmake (ffmpeg, mplayer, mariadb, etc.), it seems cmake is working ok.  
I make sure gnuradio is building 'clean', ie, I run git clean -d -x -f  as part of the build.
I installed other python components as follows:
pip install cheetah sphinx lxml numpy scipy
(other dependencies were met with this, but not numpy).  
Is there a specific version of numpy that I need?  Currently my script says I'm running
version 1.15.3  (which is the stock version for linux mint 19).  To be complete, the version of boost
related to this system are:
dpkg -l | grep numpy
ii  libboost-numpy-dev                         1.65.1.0ubuntu1                     amd64        Boost.Python NumPy extensions development files (default version)
ii  libboost-numpy1.65-dev                     1.65.1+dfsg-0ubuntu5                amd64        Boost.Python NumPy extensions development files
ii  libboost-numpy1.65.1                       1.65.1+dfsg-0ubuntu5                amd64        Boost.Python NumPy extensions
ii  python3-numpy                              1:1.13.3-2ubuntu1                   amd64        Fast array facility to the Python 3 language

Sorry for the long post, but I wanted to be complete and give as much information as I have.
Thanks in advance,
Bob

noc0lour, 2018-10-27:
Hi, thank you for the report.  In general your setup looks sane. Can you execute the cmake command and paste the logs into a `gist` or pastebin and give the link here?
Also python(2)-numpy is not showing up in `dpgk -l` you posted. Maybe numpy for python2 is in a virtualenv and not used in the build? 
In general you can try to run `cmake` with `-DPYTHON_EXECUTABLE=/usr/bin/python3` to force GR to use `python3`.

mayorbobster, 2018-10-27:
Hello, thank you for the reply.
In my build script I run:
cmake ../  (with no other parameters)
I have tried to run cmake -DENABLE_GRC=ON ../   (but it breaks right away)
The tail of the build looks like this:
[ 16%] Building CXX object gnuradio-runtime/swig/CMakeFiles/pmt_swig_gnuradio_runtime_swig_7dd5e.dir/pmt_swig_gnuradio_runtime_swig_7dd5e.cpp.o
[ 16%] Linking CXX executable pmt_swig_gnuradio_runtime_swig_7dd5e
Swig source
Deprecated command line option: -modern. This option is now always on.
swig error : Unrecognized option -modern
Use 'swig -help' for available options.
gnuradio-runtime/swig/CMakeFiles/pmt_swig_gnuradio_runtime_swig_7dd5e.dir/build.make:96: recipe for target 'gnuradio-runtime/swig/pmt_swig_gnuradio_runtime_swig_7dd5e' failed
make[2]: *** [gnuradio-runtime/swig/pmt_swig_gnuradio_runtime_swig_7dd5e] Error 1
make[2]: *** Deleting file 'gnuradio-runtime/swig/pmt_swig_gnuradio_runtime_swig_7dd5e'
CMakeFiles/Makefile2:1421: recipe for target 'gnuradio-runtime/swig/CMakeFiles/pmt_swig_gnuradio_runtime_swig_7dd5e.dir/all' failed
make[1]: *** [gnuradio-runtime/swig/CMakeFiles/pmt_swig_gnuradio_runtime_swig_7dd5e.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2

... It looks like volk is failing, but when I install/update, I include recursive parts from git, ie: some of my build script looks like:
#!/bin/bash

SD=`pwd`/gnuradio.git

# CORES=`cat /proc/cpuinfo | grep siblings | tail -1 | tr -s ' ' | sed 's,.*: ,,'`
CORES=`nproc`
CORES=`expr $CORES + $CORES / 2`
# note std=c++11 replaces c++0x
# export CPPFLAGS="-std=c++11"

if [ ! -d $SD ]
then
git clone --recursive https://github.com/gnuradio/gnuradio.git $SD
cd $SD
else
cd $SD
git pull
git submodule update --recursive
fi

# ./configure
if [ -d build ]
then
rm -rf ./build
mkdir build
cd build
else
mkdir build
cd build
fi

And (I think) I've created a public gist file at address https://gist.github.com/mayorbobster/e824e6fa4ec489129360f1109ff28eba.js

mayorbobster, 2018-10-27:
... Gist didn't work so well, here is CMakeError.log

[CMakeError.log](https://github.com/gnuradio/gnuradio/files/2522325/CMakeError.log)
[CMakeOutput.log](https://github.com/gnuradio/gnuradio/files/2522326/CMakeOutput.log)

mayorbobster, 2018-12-15:
Note: to anyone reading this: ultimately fixed by forcing python version in cmake, ie:
cmake   -DPYTHON_EXECUTABLE=/usr/bin/python ../
which picks up python 2.7 instead of python 3+, numpy is happy, lxml is happy, gnuradio-companion works...

closed 2018-12-15 by mayorbobster

robincoxe, 2019-01-05:
I ran into this issue as well on Ubuntu 18.04.  The  (non-obvious) culprit appears to be the presence of libopenblas, which prevents numpy from importing properly.  `sudo apt-get remove libopenblas-base` did the trick.

mayorbobster, 2019-01-07:
I just caught up with this.  I *do* install openblas, from github as part of the whole build process. 
One of my scripts looks suspiciously like:
 #!/bin/bash
RUNDIR=`pwd`
SD=blas.git
CORES=`nproc`
export FORTRAN=`which fortran`
if [ ! -d $SD ]
then
git clone https://github.com/xianyi/OpenBLAS.git $SD
cd $SD
else
cd $SD
git pull
fi
if [ -d build ]
then
rm -rf ./build
fi
mkdir build
cd build
make clean
cmake -DBUILD_SHARED_LIBS:BOOL=ON ..
make -j $CORES
# make test
make install

ldconfig
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/lib/libblas

... so I build openblas as part of the whole thing.  And it runs, but I have to use
 cmake   -DPYTHON_EXECUTABLE=/usr/bin/python ../  
when building gnuradio, and `which python` is also /usr/bin/python, and it happens to be 
Python 2.7.15rc1 (default, Nov 12 2018, 14:31:15)
I didn't suspect openblas but I build it along with fftw, atlas, gmp, gsl,  mpir and numpy since they are all math libraries I use.  (all from github sources).