gnuradio/gnuradio #2121: New Depreciation Warnings after last fixes

dl1ksv, 2018-11-01:
/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/analog/fm_emph.py:108: DeprecationWarning: invalid escape sequence \/
  """
/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/analog/fm_emph.py:257: DeprecationWarning: invalid escape sequence \/
  """

btashton, 2018-11-01:
I'll take this.  It is a trivial fix, but there are actually a bunch more locations that need to get fixed up.

@marcusmueller would you be open to adding to the CI flow a check using `pycodestyle` perhaps starting with white-listing error/warning codes (there are so many PEP8 failures that a black-list approach is likely out of the question at this point).
To detect most of these errors after a build I run

```
$ pycodestyle-3 --select=W605 ./
./volk/python/volk_modtool/cfg.py:34:10: W605 invalid escape sequence '\$'
./volk/python/volk_modtool/cfg.py:34:10: W605 invalid escape sequence '\$'
./volk/python/volk_modtool/volk_modtool_generate.py:36:11: W605 invalid escape sequence '\s'
./volk/python/volk_modtool/volk_modtool_generate.py:37:11: W605 invalid escape sequence '\s'
./volk/python/volk_modtool/volk_modtool_generate.py:38:11: W605 invalid escape sequence '\s'
./volk/python/volk_modtool/volk_modtool_generate.py:38:14: W605 invalid escape sequence '\('
./volk/python/volk_modtool/volk_modtool_generate.py:39:11: W605 invalid escape sequence '\s'
./volk/python/volk_modtool/volk_modtool_generate.py:41:10: W605 invalid escape sequence '\s'
./volk/python/volk_modtool/volk_modtool_generate.py:41:22: W605 invalid escape sequence '\['
./volk/python/volk_modtool/volk_modtool_generate.py:41:28: W605 invalid escape sequence '\]'
./volk/python/volk_modtool/volk_modtool_generate.py:42:11: W605 invalid escape sequence '\s'
./volk/python/volk_modtool/volk_modtool_generate.py:42:20: W605 invalid escape sequence '\('
./volk/python/volk_modtool/volk_modtool_generate.py:42:27: W605 invalid escape sequence '\['
./volk/python/volk_modtool/volk_modtool_generate.py:42:30: W605 invalid escape sequence '\]'
./volk/python/volk_modtool/volk_modtool_generate.py:44:11: W605 invalid escape sequence '\s'
./volk/python/volk_modtool/volk_modtool_generate.py:44:25: W605 invalid escape sequence '\('
./volk/python/volk_modtool/volk_modtool_generate.py:44:38: W605 invalid escape sequence '\('
./volk/python/volk_modtool/volk_modtool_generate.py:44:40: W605 invalid escape sequence '\)'
./volk/python/volk_modtool/volk_modtool_generate.py:44:42: W605 invalid escape sequence '\)'
./volk/python/volk_modtool/volk_modtool_generate.py:74:16: W605 invalid escape sequence '\.'
./volk/python/volk_modtool/volk_modtool_generate.py:88:27: W605 invalid escape sequence '\.'
./volk/python/volk_modtool/volk_modtool_generate.py:187:11: W605 invalid escape sequence '\s'
./volk/python/volk_modtool/volk_modtool_generate.py:188:11: W605 invalid escape sequence '\s'
./volk/python/volk_modtool/volk_modtool_generate.py:255:11: W605 invalid escape sequence '\s'
./volk/python/volk_modtool/volk_modtool_generate.py:256:11: W605 invalid escape sequence '\s'
./volk/python/volk_modtool/volk_modtool_generate.py:303:14: W605 invalid escape sequence '\s'
./volk/python/volk_modtool/volk_modtool_generate.py:303:14: W605 invalid escape sequence '\s'
./volk/gen/volk_kernel_defs.py:56:12: W605 invalid escape sequence '\s'
./volk/gen/volk_kernel_defs.py:56:18: W605 invalid escape sequence '\s'
./volk/gen/volk_kernel_defs.py:56:23: W605 invalid escape sequence '\w'
./volk/gen/volk_kernel_defs.py:124:19: W605 invalid escape sequence '\w'
./volk/gen/volk_kernel_defs.py:136:19: W605 invalid escape sequence '\s'
./volk/gen/volk_kernel_defs.py:136:29: W605 invalid escape sequence '\s'
./volk/gen/volk_kernel_defs.py:136:33: W605 invalid escape sequence '\w'
./volk/gen/volk_kernel_defs.py:136:37: W605 invalid escape sequence '\s'
./volk/gen/volk_kernel_defs.py:156:19: W605 invalid escape sequence '\w'
./build/get_swig_deps.py:5:19: W605 invalid escape sequence '\s'
./build/get_swig_deps.py:6:12: W605 invalid escape sequence '\s'
./gr-analog/python/analog/fm_emph.py:108:-2261: W605 invalid escape sequence '\/'
./gr-analog/python/analog/fm_emph.py:108:-2259: W605 invalid escape sequence '\/'
./gr-analog/python/analog/fm_emph.py:108:-2257: W605 invalid escape sequence '\/'
./gr-analog/python/analog/fm_emph.py:257:-3183: W605 invalid escape sequence '\/'
./gr-analog/python/analog/fm_emph.py:257:-3181: W605 invalid escape sequence '\/'
./gr-analog/python/analog/fm_emph.py:257:-3179: W605 invalid escape sequence '\/'
./gr-analog/python/analog/fm_emph.py:257:-3107: W605 invalid escape sequence '\ '
```

marcusmueller, 2018-11-02:
@btashton I'm open to anything that improves code quality :)

Our QA / testing infra guru is @noc0lour of GNU Radio and Buildbot fame: @noc0lour, how can we most elegantly integrate such overall code checking into the CI? Add a tool to `dtools/`, so that users can run it pre-push in the future?

PR #2123 (2018-11-12): Use raw strings for regexp and docstrings that have desired backslashes
Resolves #2121 DeprecationWarning: invalid escape sequence /
commit 67371e79 (2018-11-12): Use raw strings for regexp and docstrings that have desired backslashes

Resolves #2121 DeprecationWarning: invalid escape sequence /
closed 2018-11-12 by mbr0wn