gnuradio/gnuradio #2134: Document usage of symbol_sync

JayKickliter, 2018-11-03:
I got really excited after watching @awalls-cx18's talk from last year on his new `symbol_sync_xx` blocks, but it faded pretty quickly. I can live with undocumented function parameters (which is par for the course in GNU Radio), but `symbol_sync` has no examples showing how to integrate it into a receiver. All I see is a flowgraph that plots the time-domain output. What should go before/after the block? Or, how do I turn RF samples into bits? I realize this is an advanced block with complicated knobs, but a starting point would be nice.

This being open-source, I'm more than happy to write the documentation/examples if someone can nudge me in the right direction.

777arc, 2018-11-04:
Sry didn't mean to close.  If someone writes a guide about the params and how to use it, I'll help put it in the right format and get it added in the right spot.

awalls-cx18, 2018-11-04:
On November 3, 2018 6:02:52 PM EDT, Jay Kickliter <notifications@github.com> wrote:
> I got really excited after watching @awalls-cx18's talk from last year
> on his new `symbol_sync_xx` blocks, but it faded pretty quickly. I can
> live with undocumented function parameters (which is par [...]

So you have to limit the scope here.  This is not really the place to into,how to design an entire receiver.

This is just a single synchronization function in a receiver that should be implementing multiple synchronization functions.  Any synchronization function performs one or more of filtering, estimation, correction, and tracking.

It would be best to just document, in plain language, what the block does and how it expects its inputs conditioned.  (I'll write up a succinct synopsis tommorrow, upon which someone else can then expound.)

Anything more and you'll be trying to document how to build a full receiver - which is better left to textbooks.

777arc, 2018-11-04:
@awalls-cx18 I'll gladly incorporate your synopsis into the docs.  Feel free to just paste it in this issue thread.

777arc, 2018-11-29:
@awalls-cx18 Is it possible to write up a brief synopsis of this block?  I can act as the editor.

awalls-cx18, 2018-11-29:
On November 29, 2018 12:51:18 PM EST, Marc L <notifications@github.com> wrote:
>@awalls-cx18 Is it possible to write up a brief synopsis of this block?
> I can act as the editor. 

Yes.  Holiday travel and work got in the way.  I'll write something up this weekend.  I swear...

777arc, 2019-01-17:
@awalls-cx18 ping to see if you are still interested in providing a blurb of docs that ill merge in for you

awalls-cx18, 2019-01-22:
@777arc echo I keep intending to do so, but my track record on actually getting it done appears to be very poor. :(  All I would really be doing is massaging some of the content from my GRCon17 brief and maybe expounding on some things with a full sentence or two. 

The content I would massage into documentation is: 
https://www.gnuradio.org/wp-content/uploads/2017/12/Andy-Walls-Samples-to-Digital-Symbols.pdf
Slide 5
Bottom of Slide 6
Top of Slide 7
Slide 8
Slide 9
Top of Slide 14
Slide 15
Slide 23
Slide 25
Slides 34-36
Slide 38

If you really want to cook book it down:

The block outputs a sample stream synchronized with the center of data symbols received on the input sample stream.

1. Condition your inputs into the block: Channel filter/bandlimit, AGC, Level shift, and matched filter (if you use the MMSE interpolation filter and thus matched filtering not done in the block).  The block expects the input symbols to be peaked at symbol centers after matched filtering, except when using the MSK TEDs.

2. Select a TED.  Perform Modeling, Simulation, and Analysis to determine the expected TED gain for your expected input conditions.

3. Select a damping factor of 0.707 or 1.0 or higher.  Usually 1.0 is preferred.

4. Select a small Normalized Loop Bandwidth (Normalized Natural Frequency).  This parameter is usually much, much smaller than pi, but greater than 0.

5. Perform Modeling, Simulation, and Analysis to choose the optimal values for damping factor and Normalized loop bandwidth, or run a data stream through the block in GNURadio and tweak the parameters on the fly to find parameters that appear to work well enough.

6. This is a PLL based symbol synchronizer.  There will be a trade off between initial synchronization acquisition performance and symbol tracking stability.  A time_est tag or clock_est tag placed on an the input sample stream, say by a preamble correlator block, can be used to speed initial synchronization acquisition, while using parameters that would otherwise mean acquiring initial symchronization slowly..

777arc, 2019-01-22:
Awesome, this will work, I'll incorporate it in!

777arc, 2019-02-26:
@awalls-cx18 Turns out that link is dead, any chance you can email me the file?  MarcLL at vt.edu

awalls-cx18, 2019-02-27:
@777arc : It turns out that @noc0lour is bringing the GRCon 2017 briefs back to the website:
https://www.gnuradio.org/grcon/grcon17/presentations/

He has Monday's briefs done.  I think mine was on Wednesday.

Anyway, I'll attach it here.
[Samples_to_Digital_Symbol_AWalls_20170912a.pdf](https://github.com/gnuradio/gnuradio/files/2911247/Samples_to_Digital_Symbol_AWalls_20170912a.pdf)

777arc, 2019-03-02:
Holy crap I didn't realize how messy the current symbol_sync docs are, just looking through the include files makes my head explode.  So much repetition and bad formatting.  I think most of it is targeted towards devs, and I don't want to remove any of that stuff, so for now I have attempted to make a decent description of the block/parameters in [this wiki page](https://wiki.gnuradio.org/index.php/Symbol_Sync).  These wiki pages tend to get indexed by google pretty quickly, so if a GR user googles the name of the block and "gnuradio" it will come up near the top.

777arc, 2019-03-02:
@awalls-cx18  is there an example flowgraph using this block that you think would be good to point out?

awalls-cx18, 2019-03-03:
On March 1, 2019 10:27:11 PM EST, Marc L <notifications@github.com> wrote:
>@awalls-cx18  is there an example flowgraph using this block that you
>think would be good to point out?

As noted in the final 2-3 slides of the presentation, there are some examples in gnuradio/gr-digital/examples/demod/symbol_sync*grc.

They get installed under (prefix)/share/gnuradio/examples/digital/demod, IIRC.  Please check the path to be sure.

777arc, 2019-03-03:
@awalls-cx18 To me, those two flowgraphs just feel like flowgraphs meant to "test" the symbol_sync block.  I was thinking more along the lines of where this block would be placed in something like a receiver meant to sync to BPSK received over the air.  Let me ask you this- if I just replaced the Polyphase Clock Sync block in our popular [PSK demod tutorial](https://wiki.gnuradio.org/index.php/Guided_Tutorial_PSK_Demodulation) (scroll all the way to the bottom for the final flowgraph), using your guide in the slides about how to translate args between the two blocks, would that be a reasonable "example flowgraph" to provide to users?

awalls-cx18, 2019-03-03:
@777arc Those two flowgraphs are not meant to test the block, but are meant to be instructive: to give the user a feel for how the block behaves with various settings.  In those two flowgraphs, the user can change the Expected TED gain, damping factor, and loop bandwidth on the fly and watch how the block converges by monitoring the Instantaneous Symbol Period and Avergae Symbol Period outputs.  The user can turn on a time_est tag and see how that speeds acquisition of timing at the start of a burst.  The user can turn on just a simple test tag to see how the block propagates tags properly.  The user can change the signal on the fly between packet, high freq (clock pattern), low freq, and pulse train.  Th M&M TED in particular has it's gain collapse on a clock pattern signal, and the user can see the practical implication of that using these flowgraphs, for example.  The flowgraphs also demonstrate the use of a matched filter to peak the symbol centers.  I chose a simple baseband PAM signal, so the user wouldn't get hung up on the modulation, but instead focus on the symbol centers and symbol synchronization.

To answer your other question.  Sure you can place this block in the PSK tutorial, as it is a drop in replacement for the PFB clock sync block, but you'll need to tweak parameters a little.  The PFB clock sync block assumes the expected TED gain is always 1.0 (when that is rarely the case).  The PFB clock sync block is also set by default to be ridiculously over-dampedt:  https://github.com/gnuradio/gnuradio/blob/master/gr-digital/lib/pfb_clock_sync_ccf_impl.cc#L81  (the comment there is a blatant  falsehood.)

Personally, I don't find showing the symbol sync block working for a continuous stream BPSK tutorial flowgraph to be instructive on how to use the symbol sync block itself.  It's use in the tutorial will have the effect of encouraging people to cut and paste it, parameters and all, into their working context, for which the parameters in the tutorial may not work very well.  It does however show the block in the context of a simple, continuous symbol stream demodulator.  Which is OK, if you are trying to teach someone how to build a simple receiver for non-burst data.

Honestly, I would like to see all the tutorials updated to the new symbol sync block, so people stop asking about the PFB clock sync block and M&M block, since the old blocks have various bugs.

duggabe, 2023-07-14:
Since most of the issues raised here have been covered, the Clock Recovery MM and Polyphase Clock Sync blocks are now deprecated, and the issue is over four years old, I suggest the issue to be closed.

closed 2023-07-15 by JayKickliter