gnuradio/gnuradio #2142: Py3 doesn't have `iteritems`, but we do (in a lot of places)

marcusmueller, 2018-11-04:
as phansel raised in PR #2133, this breaks in Py3. By sheer luck we've dodged this bullet so far.

Need to fix those; in most cases, just replacing `iteritems` with `items` will work, since **both**

* the (py2) usage doesn't care whether it's actually an iterator; `items` gives an iterable, just as much **and**
* the dictionary iterated about is small enough for us to ignore the list-generation overhead.

So, fix is easy, but must be done with a tiny bit of respect for the context in which `iteritems` was used.

marcusmueller, 2018-11-04:
`ag -l --python iteritems` delivers:

<pre>
grc/core/io/yaml.py
grc/core/utils/extract_docs.py
grc/core/utils/expr_utils.py
grc/core/ParseXML.py
grc/core/schema_checker/validator.py
grc/core/generator/hier_block.py
grc/core/params/param.py
grc/core/platform.py
grc/core/blocks/block.py
grc/core/cache.py
grc/tests/resources/file3.block.yml
grc/gui/PropsDialog.py
grc/gui/ParserErrorsDialog.py
grc/gui/canvas/block.py
grc/gui/BlockTreeWindow.py
grc/converter/main.py
volk/gen/volk_machine_defs.py
volk/gen/volk_arch_defs.py
gnuradio-runtime/python/gnuradio/ctrlport/gr-perf-monitorx
</pre>

Since VOLK is involved: pinging @n-west .

skoslowski, 2018-11-04:
Please read https://pythonhosted.org/six/#six.iteritems

marcusmueller, 2018-11-04:
I've done so; there's still these two aforementioned non-six occurrences, hence the issue.

btashton, 2018-11-04:
@marcusmueller I'm trying to understand the scope of what you are asking for here.  There are a few places were we use `six.iteritems()` where `items()` would be fine are you wanting these changed?

Outside of that there are only 3 occurrences that NEED to be fixed:

- [ ] ./gnuradio-runtime/python/gnuradio/ctrlport/gr-perf-monitorx:406:                sorted_fullness = sorted(blockport_fullness.iteritems(),
- [ ] ./gnuradio-runtime/python/gnuradio/ctrlport/gr-perf-monitorx:460:                sorted_work = sorted(work_times.iteritems(), key=operator.itemgetter(1))
- [ ] ./grc/tests/resources/file3.block.yml:53:        self._${id}_choices_inv = dict((v,k) for k,v in self._${id}_choices.iteritems())