gnuradio/gnuradio #2145: `gr_plot_*`: depends on PyQt4

marcusmueller, 2018-11-04:
These uitilities depend on a version of Qt that might be gone soon, and we're dropping Qt4 as primary requirement for gr-qtplot (Qt5 works just as well).

So, we need Qt5-compatible versions of these programs.

These shouldn't be impossible to write. If we want to include `pyqtgraph` as a dependency, we could do 

```python
#!/usr/bin/env python
from pyqtgraph.Qt import QtGui, QtCore
import numpy
import pyqtgraph as pg

def main(data, name = None):
    app = QtGui.QApplication([])
    win = pg.GraphicsWindow(title="Plotting")
    win.setWindowTitle("Plotting" if name is None else "Plotting " +  name)
    pg.setConfigOptions(antialias=True)

    p1 = win.addPlot(title="Plotting Real (red), Imag (green), angle (blue)")
    p1.addLegend()

    p1.plot(data.real, pen=(255,0,0), name="real")
    p1.plot(data.imag, pen=(0,255,0), name="imag")
    # could enable on command line flag: numpy.arctan2 is slow
    # hm. Maybe use VOLK instead.
    #p1.plot(numpy.arctan2(data.imag, data.real), pen=(0,0,255))

    QtGui.QApplication.instance().exec_()

if __name__ == '__main__':
    import sys
    import argparse
    parser = argparse.ArgumentParser()
    parser.add_argument("FILE", type=argparse.FileType("r"))
    args = parser.parse_args()
    data = numpy.fromfile(args.FILE, dtype=numpy.complex64)
    main(data)
```

which actually is much much faster than the current matplotlib-based implementations, but so far lacks the full feature set of `gr_plot_iq`:

- [x] display complex data
- [x] zoom
- [ ] specify type on command line
- [ ] specify sampling rate
- [ ] specify display block size
- [ ] specify display block start
- [ ] help / man page

Specifically, we would need

* `gr_plot_iq`,
*  `gr_spectrogram_plot`,
* `gr_constellation_plot`, (already gr-qtgui-based, so probably just modify the `import`s)
* `gr_plot_fft`,
* `gr_psd_plot`

I'd personally would be willing to discuss whether pyqtgraph or matplotlib is the worse dependency to have.

btashton, 2018-11-15:
@mbr0wn In master you got rid of a lot of the unnecessary plotting utilities.  Could we just consolidate the remaining ones to just `gr_plot` and then take command line arguments to further refine the plots.
Having `gr_plot_fft`, `gr_plot_fft_f`, `gr_plot_fft_c` seems silly.  I would even go further and make `fft` `psd` options off of `gr_plot`

btashton, 2018-11-15:
disregard that, I see that these were part of that.

btashton, 2018-12-14:
I think the only thing from this issue that is missing from `grplot` is selecting a partial block of the file.  The underlying code should make this fairly easy implement, but I need to think about what the UI might look like.

closed 2021-03-24 by marcusmueller