gnuradio/gnuradio #2154: CMake enables gr-uhd but compilation fails due to missing Python module mako

rear1019, 2018-11-06:
On a system without `python-mako` CMake enables `gr-uhd`, however compilation fails due to missing Python module `mako`. I can reproduce this error on Ubuntu 18.04 LTS and Arch Linux.

noc0lour, 2018-11-09:
Hi @rear1019 which version of GNU Radio and UHD are you trying to build?

rear1019, 2018-11-09:
Ubuntu 18.04 LTS: GNU Radio git revision 97903bc3d, libuhd-dev 3.10.3.0 (from Ubuntu’s repository)
Arch Linux: GNU Radio git revision 5823f31 and 97903bc3d, libuhd 3.13.0.1 (from Arch Linux’ repository)
Looking more carefully at CMakes’s output I have discovered that CMake actually detects that `mako` ist missing:
```
-- Configuring gr-uhd support...
--   Dependency Boost_FOUND = 1
--   Dependency UHD_FOUND = TRUE
--   Dependency ENABLE_GNURADIO_RUNTIME = ON
--   Dependency ENABLE_GR_FILTER = ON
--   Dependency ENABLE_GR_BLOCKS = ON
--   Dependency ENABLE_GR_ANALOG = ON
--   Enabling gr-uhd support.
--   Override with -DENABLE_GR_UHD=ON/OFF
--   UHD Version: 3.13.0.1-0-unknown
-- Python checking for Mako >= 0.4.2 - not found
```

noc0lour, 2018-11-09:
Hm, maybe we need to manually add some required flag there, too.

marcusmueller, 2018-11-09:
Might already be solved in UHD itself. Mbr0wn?

noc0lour, 2018-11-09:
GNU Radio is not calling UHDs `CMakeLists.txt` and shouldn't care about any Python dependencies UHD has. The fact that cmake is not stopping if `mako` is not found is a problem. Since it's a build dependency for UHD blocks (and for volk).

mbr0wn, 2018-12-15:
Gr-uhd uses mako during build time.

mbr0wn, 2018-12-16:
So here's what's going on: You can build GNU Radio without mako, you just don't get GRC in that case. However, the grc/ subdirectories of all components get touched even if GRC is disabled. gr-uhd/grc assumes that mako is there, because GRC does.

closed 2018-12-18 by mbr0wn

rsandler00, 2021-02-04:
`pip install mako` solved this for me

PR #2240 (2018-12-18): cmake: Only go into grc/ subdirs when ENABLE_GRC=ON
Fixes #2154.