gnuradio/gnuradio #2159: Update the sps on symbol sync block

marcosvpinho, 2018-11-11:
I'm using symbol sync block (symbol_sync_ff) on my project and i use many different sps values that i need reset on runtime.

I'm trying to reset sending clock_est tag. Is this the right way to reset the sps value?

key = pmt.intern("clock_est")
value = pmt.to_pmt((0.0, 45))

marcosvpinho, 2018-11-16:
@awalls-cx18

closed 2018-11-18 by marcosvpinho

awalls-cx18, 2018-11-18:
On November 18, 2018 12:10:43 PM EST, marcosvpinho <notifications@github.com> wrote:
>Closed #2159.

The clock_est tag resets the internal estimate of the sps in the block to aid in faster acquisition.  The sps range that the block allows is the nominal value +\- the user specified allowed deviation.  Switching between multiple, very different sps values can't be done via the clock_est tag.