gnuradio/gnuradio #2160: grc:does not correcly  builds dummy blocks

dl1ksv, 2018-11-12:
If a flowgraph contains a block without a block description ( perhaps depreceated in 3.7 and now missing , or missing from other reasons ) grc should build a dummy block. https://github.com/gnuradio/gnuradio/blob/b1e0b54d02bb981eba734ea29174e6134950dde6/grc/core/FlowGraph.py#L371

But the dummy block will not be build:

    Loading: "/home/schroer/gnuradiocomponents/gr3.8tests/OptionalPadTest.grc"
    Block id "hier2Msg" not found
    Error: 'id'
>>> Failure
Traceback (most recent call last):
  File "/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/gui/MainWindow.py", line 261, in new_page
    file_path=file_path,
  File "/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/gui/Notebook.py", line 107, in __init__
    flow_graph.import_data(initial_state)
  File "/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/core/FlowGraph.py", line 377, in import_data
    block.import_data(**block_data)
  File "/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/core/blocks/block.py", line 344, in import_data
    self.params['id'].value = name
KeyError: 'id'

and in consequence, the flowgraph will not be build.
The fowgraph isn't   build even if the missing block is disabled.
So you have no chance to edit this  flowgraph and make it work again.

The only way to get this flowgraph building is to modify the grc file by hand.

haakov, 2018-11-14:
I suspect the fix for this will be similar to the fix for the virtual blocks :)

noc0lour, 2019-03-13:
Merged in 88a1549f4e69fefd4f02eb826ada04cbd36b0849

closed 2019-03-13 by noc0lour