gnuradio/gnuradio #2173: python: Use a python logger for the logging in utilities

btashton, 2018-11-15:
Right now there is no standard logging interface being used across the python utilities.  This results in a mixture of no logging and print statement logging, both of which are not ideal.  I would also advocate for adding a standard argparse argument for the utilities that directly controls the log level.

michaelld, 2018-11-15:
Even better would be to use the same logger for both C++ and Python! Or, at least have a common "print mutex" to keep printing in various domains from muxing & becoming difficult to parse. I have no idea if such a logger exists; having a C++ based print mutex & then using SWIG to provide a function to lock & unlock the mutex from Python would be quite doable.

btashton, 2018-11-15:
Python could hook into such an interface fairly easily by just registering a custom log handler. Each side would then have to have a logging thread (which would be better anyway).  So on the python side for now we would just want to start using the standard logging interfaces in preparation.

noc0lour, 2018-11-15:
So basically you'd suggest piping all logs through the available GR C++ interface? And having a central knob there to configure the levels/output/etc ?

michaelld, 2018-11-15:
yes!

btashton, 2018-11-15:
@noc0lour No I would not want to do that. I would like to have only the mutex in there.  Any external logging library could then say hey I want access to stderr to write some logs. In python we would then have `class GRFriendlyStreamLogHandler()` that would buffer the log messages and ask for the mutex to write to the stream. If we really wanted to I could see dropping the python thread and just sending them to a C++ log handler, but I would really want to keep the python interface as just registering a handler.

noc0lour, 2018-11-15:
Oh, I see. I like the python logging suggestion either way.

michaelld, 2018-11-15:
Oh sorry I misinterpreted what you wrote, @noc0lour  ... I prefer the mutex version too: request it to print, print, then release it & let the next one print. Not as fast as using a FIFO queue to hold the messages to print, but probably easier to implement than such a single printing interface.

btashton, 2018-11-15:
handlers can have their own log level set, so we probably could could make that a two way interface.  I guess my point is once we are using a unified handler on the python side we can do all of these options without changing much

NeuroForLunch, 2021-06-24:
3.9 has a logger. Please close this issue.

marcusmueller, 2023-12-22:
actually using the existing logger, whether it be the python logging module or directly the GR logging, is still an outstanding enhancement