gnuradio/gnuradio #2178: Flowgraph Errors

dl1ksv, 2018-11-16:
commit https://github.com/gnuradio/gnuradio/commit/d23fa96427c00bc2d149e6c55cf1f43ac956c213 produces a lot of Flowgraph errors  when starting grc and loading hier2 blocks that where generated by grc.

    "/home/schroer/gnuradiocomponents/gr3.8tests/OptionalPadTest.grc"
    ERROR:gnuradio.grc.core.FlowGraph:Failed to evaluate expression in namespace
    Traceback (most recent call last):
      File "/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/core/FlowGraph.py", line 193, in renew_namespace
    exec(expr, namespace)
      File "<string>", line 1, in <module>
    ModuleNotFoundError: No module named 'msgporttest'
    ERROR:gnuradio.grc.core.FlowGraph:Failed to evaluate expression in namespace
      Traceback (most recent call last):
      File "/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/core/FlowGraph.py", line 193, in renew_namespace
    exec(expr, namespace)
      File "<string>", line 1, in <module>
     ModuleNotFoundError: No module named 'msgporttest'
    >>> Done

and so on.

I think this behaviour is intended. But the messages are misleading. As the flowgraph works as before the commit it's not an error, so a warning instead of an error should be reported.

Which exceptions are real errors and which not ?
And what should user do who come across such reports ?

btashton, 2018-11-16:
This is exactly why I added that. I expect that we will see more of these errors being logged. I would say that it is an error not a warning we explicitly asked it to evaluate something and it could not find the module to do so.

dl1ksv, 2018-11-16:
I disagree, as the flowgraph works. But it's not worth to discuss it if it's an error or not. 

The main question is: What is the solution ? Is the check unneeded , or what kind of exceptions are realy faulty ? Or did grc generate wrong  ( but working ) code ? 

If you expect more such messages you should users give a hint what to do with such messages.

btashton, 2018-11-16:
Before we were silently covering up a large swath of possible errors, we are now logging the exception, but in this case we do not have much context. Do you have an example flow graph I can execute to see this? I'm happy to look into the actual bug.

dl1ksv, 2018-11-16:
You need the latest master and #1925  merged to make the flowgraph run

**grc output** 

 grc
DEBUG:grc:Loading platform
DEBUG:grc:Loading application
DEBUG:grc:Running
<<< Welcome to GNU Radio Companion 3.8tech-preview-122-g167c3975 >>>

Block paths:
	/home/schroer/.grc_gnuradio
	/usr/local/gnuradio-volker3.8/share/gnuradio/grc/blocks

Loading: "/home/schroer/gnuradiocomponents/gr3.8tests/OptionalPadTest.grc"
ERROR:gnuradio.grc.core.FlowGraph:Failed to evaluate expression in namespace
Traceback (most recent call last):
  File "/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/core/FlowGraph.py", line 193, in renew_namespace
    exec(expr, namespace)
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'msgporttest'
ERROR:gnuradio.grc.core.FlowGraph:Failed to evaluate expression in namespace
Traceback (most recent call last):
  File "/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/core/FlowGraph.py", line 193, in renew_namespace
    exec(expr, namespace)
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'msgporttest'
>>> Done
ERROR:gnuradio.grc.core.FlowGraph:Failed to evaluate expression in namespace
Traceback (most recent call last):
  File "/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/core/FlowGraph.py", line 193, in renew_namespace
    exec(expr, namespace)
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'msgporttest'

Loading: "/home/schroer/gnuradiocomponents/gr3.8tests/Hier2Msg.grc"
>>> Done

(gnuradio-companion:11384): Gtk-CRITICAL **: 17:50:49.199: gtk_main_quit: assertion 'main_loops != NULL' failed

[OptionalPadTest.grc.gz](https://github.com/gnuradio/gnuradio/files/2590214/OptionalPadTest.grc.gz)
[Hier2Msg.grc.gz](https://github.com/gnuradio/gnuradio/files/2590216/Hier2Msg.grc.gz)

btashton, 2018-11-16:
I was able to reproduce and I opened a ticket specifically for this  #2183  The reason you see the flow graph still work correctly is this is only impacting the gui not the generated flow graph code.   Are thee any other exceptions you are seeing captured here?

btashton, 2018-11-16:
@dl1ksv can you try #2183

dl1ksv, 2018-11-16:
@btashton I tried #2185.

The error messages disappeared. :+1:  but the disabled blocks are not displayed.
After switching between patched and non patched version the disabled blocks are displayed.
No idea why.

Attached the correct flowgraph
[OptionalPadTest.pdf](https://github.com/gnuradio/gnuradio/files/2590812/OptionalPadTest.pdf)
 

At the moment I see no further execption messages, but I'll open a new issue if I get one.

btashton, 2018-11-16:
To clarify after you swapped between the two versions the disable blocks showed up in both versions?  I was not able to reproduce that.  I did get a template exception, but I have not been able to reproduce that now...

dl1ksv, 2018-11-16:
Yes, you are right. It's very strange. At the moment I can' reproduce this behaviour. I'll go and check some more hier blocks, to see if it works.

btashton, 2018-11-16:
I'm wondering if it is this issue #2186 which might also be #2045

dl1ksv, 2018-11-16:
I checked some more flowgraphs, but everthing seems to work. I'll continue tomorrow.

dl1ksv, 2018-11-17:
That the disabled blocks are not displayed sometimes depends on an incorrect handling of toggle visibility.

See #2189

dl1ksv, 2018-11-20:
A temporary fix ist to add ~/.grc_gnuradiocto to PYTHONPATH

closed 2018-11-20 by dl1ksv

RadioFreakk, 2024-08-19:
Hi there,
I've followed this repository (https://github.com/tapparelj/gr-lora_sdr/tree/master?tab=readme-ov-file) and ended up with a Transmission and Reception flowgraph, but when I try to run the flowgraph it gives me this output:
<<< Welcome to GNU Radio Companion 3.10.1.1 >>>

Block paths:
	/usr/share/gnuradio/grc/blocks
	/usr/local/share/gnuradio/grc/blocks

Loading: "/home/enb/simpletransmitter.grc"
>>> Done

Loading: "/home/enb/Downloads/reciever(2).grc"
>>> Done

Loading: "/home/enb/Downloads/lora_TX.grc"
>>> Done

Loading: "/home/enb/Downloads/lora_RX.grc"
>>> Done

Generating: '/home/enb/Downloads/lora_RX.py'

Executing: /usr/bin/x-terminal-emulator -e /usr/bin/python3 -u /home/enb/Downloads/lora_RX.py

>>> Done (return code 1)

I don't know where I'm going wrong.
Kindly someone help!!