gnuradio/gnuradio #2186: TemplateError, not all arguments converted during string formatting

btashton, 2018-11-16:
I am not able to reproduce, but I think this was what was reported here #2045
@haakov does this seem like what you saw?
```
Loading: "/home/bashton/Documents/grc/addrtest.grc"
>>> Done
Traceback (most recent call last):
  File "/home/bashton/gr-target/lib/python3.7/dist-packages/gnuradio/grc/core/blocks/_templates.py", line 75, in render
    return template.render(**namespace)
  File "/usr/lib/python3.7/site-packages/mako/template.py", line 462, in render
    return runtime._render(self, self.callable_, args, data)
  File "/usr/lib/python3.7/site-packages/mako/runtime.py", line 838, in _render
    **_kwargs_for_callable(callable_, data))
  File "/usr/lib/python3.7/site-packages/mako/runtime.py", line 873, in _render_context
    _exec_template(inherit, lclcontext, args=args, kwargs=kwargs)
  File "/usr/lib/python3.7/site-packages/mako/runtime.py", line 899, in _exec_template
    callable_(context, *args, **kwargs)
  File "memory:0x7f00288e8da0", line 59, in render_body
TypeError: not all arguments converted during string formatting

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/bashton/gr-target/lib/python3.7/dist-packages/gnuradio/grc/gui/PropsDialog.py", line 205, in update_gui
    self._update_docs_page()
  File "/home/bashton/gr-target/lib/python3.7/dist-packages/gnuradio/grc/gui/PropsDialog.py", line 229, in _update_docs_page
    block_constructor = self._block.templates.render('make').rsplit('.', 2)[-1]
  File "/home/bashton/gr-target/lib/python3.7/dist-packages/gnuradio/grc/core/blocks/_templates.py", line 77, in render
    raise TemplateError(error, text)
gnuradio.grc.core.errors.TemplateError: (TypeError('not all arguments converted during string formatting'), '<%\n    win = \'self._%s_tool_bar\'%id\n%>\\\n${win} = Qt.QToolBar(self)\n\nif ${formatter}:\n    self._${id}_formatter = ${formatter}\nelse:\n    self._${id}_formatter = lambda x: ${type.str}(x)\n\n${win}.addWidget(Qt.QLabel(${(label if (len(label) - 2 > 0) else repr(id))} + ": "))\nself._${id}_label = Qt.QLabel(str(self._${id}_formatter(self.${id})))\nself._${id}_tool_bar.addWidget(self._${id}_label)\n${gui_hint() % win}')
```

haakov, 2018-11-19:
Yep. How did this error appear?

nickoe, 2019-03-11:
It looks like this issue went to the Abyss.

noc0lour, 2019-03-31:
I'll close this issue. reopen if it appears again.

closed 2019-03-31 by noc0lour