gnuradio/gnuradio #2191: UaUaUaUaUaUa ......

Red-Owl, 2018-11-17:
identical flowchart 
UaUaUaUaUaUaUa
3.8tech-preview-115-g48aaed52 (Python 3.6.6)

dkozel, 2018-11-18:
@Red-Owl 
Hi, We're going to need more information to determine what the problem is. When you say identical flowchart, what is it identical to? This works in an older version (which?) and now doesn't work? Do you have two pieces of hardware in your flowgraph (a radio and a soundcard)? If so, this could be the two clock problem which GNU Radio does not attempt to solve in the audio sink block.

Please give some more details about what's not working. Thanks!

dkozel, 2018-11-18:
What previous version was working for you? Was the flowgraph different or the version of GNU Radio, or both? There have not been any (at all?) changes to the audio source and sink block in the past months.

You can try increasing the amount of audio buffering using the GNU Radio config file.
http://sdrgps.blogspot.com/2016/01/gnuradio-getting-audio-to-work-in-vm.html

You can compile a flowgraph with static linking to make a standalone binary, GNU Radio Companion can generate a good starting point although you will have to modify the Makefile.

dl1ksv, 2018-11-19:
What's your audio setup ? alsa , pulseaudio ?
You could remove the audio sink to see if the problem is in the sink or source.

dkozel, 2018-11-19:
`aU` means an underrun, so it will be the Sink that is reporting the error. Likely the issue is that there is not enough buffering between the Source and Sink and possibly the audio DAC and ADC are running of separate references so there will always be a mismatch in sample rates.

closed 2018-11-24 by marcusmueller