gnuradio/gnuradio #2192: Problems compiling gnuradio 3.8 on armv8 (odroidc2)

mlasch, 2018-11-19:
Compiling master branch fails on my armv8 machine running debian stretch. It seems to be a problem with volk, however compiling volk standalone works on the same system.

```
[  1%] Building C object volk/lib/CMakeFiles/volk_obj.dir/volk_machine_neon.c.o
In file included from /home/jack/gnuradio/build/volk/lib/volk_machine_neon.c:146:0:
/home/jack/gnuradio/volk/kernels/volk/volk_32u_reverse_32u.h: In function ‘volk_32u_reverse_32u_arm’:
/home/jack/gnuradio/volk/kernels/volk/volk_32u_reverse_32u.h:375:3: warning: implicit declaration of function ‘asm’ [-Wimplicit-function-declaration]
   asm("rbit %[result], %[value]"  \
   ^
/home/jack/gnuradio/volk/kernels/volk/volk_32u_reverse_32u.h:392:9: note: in expansion of macro ‘DO_RBIT’
         DO_RBIT; DO_RBIT; DO_RBIT; DO_RBIT;
         ^~~~~~~
/home/jack/gnuradio/volk/kernels/volk/volk_32u_reverse_32u.h:376:7: error: expected ‘)’ before ‘:’ token
       : [result]"=r" (*out_ptr)   \
       ^
/home/jack/gnuradio/volk/kernels/volk/volk_32u_reverse_32u.h:392:9: note: in expansion of macro ‘DO_RBIT’
         DO_RBIT; DO_RBIT; DO_RBIT; DO_RBIT;
         ^~~~~~~
....
```

```
gcc (Debian 6.3.0-18+deb9u1) 6.3.0 20170516
commit 7e514a1d2282ca8ddb6058b106f3b381b9a74a5e
```

hcab14, 2018-11-20:
It looks like the gnu compiler extensions are not used.

Can you try recompile from scratch with the option `-DCMAKE_CXX_EXTENSIONS=ON` for cmake?

marcusmueller, 2018-12-19:
Enabling GNU Extensions is something I don't _want_ to do by default, for example because this exact armv8 target might be built with clang by default (android).

But: AFAIK, `asm` is actually ISO C standard (annex J)-specified; so. hm.

marcusmueller, 2018-12-19:
@mlasch does replacing

```
asm( "add foo 2\
         "mov foo bar" ) 
```

with 
```
__VOLK_ASM(
         "add foo 2\
         "mov foo bar" ) 
```

WITHOUT enabling GNU extensions solve the problem? In that case: you'd have fixed a GNU Radio bug!

drmpeg, 2018-12-20:
I'm having similar problems on armv7-a. It completes the build, but It ends up building a completely generic system, no FPU or NEON. I tried the patch in https://github.com/gnuradio/volk/pull/215, but it doesn't work (it detects ARMv7, but still doesn't build correctly).

This is on a BeagleBoard-X15 with AM5728 processor (Cortex-A15).

hcab14, 2018-12-20:
What is the output of cmake?

drmpeg, 2018-12-20:
Without your patch:

-- Performing Test neon_compile_result
-- Performing Test neon_compile_result - Failed
-- Compiler doesn't support NEON, Overruled arch neon
-- Compiler doesn't support NEON, Overruled arch neonv7
-- Compiler doesn't support NEON, Overruled arch neonv8
-- Available architectures: generic;hardfp;orc;norc
-- Available machines: generic_orc

With your patch, it detects arm7 properly, but this code doesn't do anything:
```
    if ("${machine_name}" STREQUAL "neon")
	  if (have_mfpu_neon)
	    set(neon_flags "${neon_flags} -mfpu=neon")
	  endif (have_mfpu_neon)
	endif()
```
I don't think there's a a machine_name called "neon". I tried "neon_orc" and that seemed to help, but the build failed. Sorry, I don't have the cmake output for that.

hcab14, 2018-12-20:
> I don't think there's a machine_name called "neon". I tried "neon_orc" and that seemed to help, but the build failed. 

I see. The changes to re-enable ARMv7 do not take into account the case when building with ORC.

For me it works with `cmake -DENABLE_ORC=OFF` on a Beaglebone.

When I last tested with ORC enabled I got test errors for some of the kernels using ORC.

mlasch, 2018-12-20:
> @mlasch does replacing
> 
> ```
> asm( "add foo 2\
>          "mov foo bar" ) 
> ```
> 
> with
> 
> ```
> __VOLK_ASM(
>          "add foo 2\
>          "mov foo bar" ) 
> ```
> 
> WITHOUT enabling GNU extensions solve the problem? In that case: you'd have fixed a GNU Radio bug!

Just tested on an odroidc2, replacing asm(...)  with VOLK_ASM() fixes the problem on armv8. The file effected is `gnuradio/volk/kernels/volk/volk_32u_reverse_32u.h`

noc0lour, 2018-12-20:
hm, probably should replace all occurences with the correct VOLK macro. Should've done that in the first place.

balister, 2018-12-20:
When native compiling, make sure your CC command turns on neon. Take a look at the native toolchain files for volk and gnuradio. I suspect Debian native cc doesn't enable neon by default since it is targeted at generic boards that may or may not support neon.

drmpeg, 2018-12-21:
Latest patch from @hcab14 works good for armv7. This is with a GNU Radio 3.8 build with the following CMake invocation:

cmake -DCMAKE_INSTALL_PREFIX=/opt/gnuradio-3.7.12git -DCMAKE_CXX_FLAGS:STRING="-march=armv7-a -mfloat-abi=hard -mfpu=neon -mtune=cortex-a15" ../

CMake output:
```
-- Performing Test neon_compile_result
-- Performing Test neon_compile_result - Success
-- Performing Test have_neonv7_result
-- Performing Test have_neonv7_result - Success
-- Performing Test have_neonv8_result
-- Performing Test have_neonv8_result - Failed
-- CPU is armv7, Overruled arch neonv8
-- Available architectures: generic;hardfp;neon;neonv7;orc;norc
-- Available machines: generic_orc;neon_orc;neonv7_hardfp_orc
-- BUILD TYPE = RELEASE
-- Base cflags = -O3 -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -Wall
-- BUILD INFO ::: generic_orc ::: GNU ::: -O3 -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -Wall 
-- BUILD INFO ::: neon_orc ::: GNU ::: -O3 -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -Wall -funsafe-math-optimizations -mfpu=neon
-- BUILD INFO ::: neonv7_hardfp_orc ::: GNU ::: -O3 -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -Wall -funsafe-math-optimizations -mfpu=neon -funsafe-math-optimizations -mfloat-abi=hard
-- Compiler Version: cc (Ubuntu/Linaro 7.3.0-27ubuntu1~18.04) 7.3.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
-- ---- Adding ASM files
-- -- Detected neon architecture; enabling ASM
-- Adding source file: /home/ubuntu/xfer/gnuradio/volk/kernels/volk/asm/neon/volk_16i_max_star_horizontal_16i.s
-- Adding source file: /home/ubuntu/xfer/gnuradio/volk/kernels/volk/asm/neon/volk_32f_s32f_multiply_32f_a_neonasm.s
-- Adding source file: /home/ubuntu/xfer/gnuradio/volk/kernels/volk/asm/neon/volk_32f_x2_add_32f_a_neonasm.s
-- Adding source file: /home/ubuntu/xfer/gnuradio/volk/kernels/volk/asm/neon/volk_32f_x2_add_32f_a_neonpipeline.s
-- Adding source file: /home/ubuntu/xfer/gnuradio/volk/kernels/volk/asm/neon/volk_32f_x2_dot_prod_32f_a_neonasm.s
-- Adding source file: /home/ubuntu/xfer/gnuradio/volk/kernels/volk/asm/neon/volk_32f_x2_dot_prod_32f_a_neonasm_opts.s
-- Adding source file: /home/ubuntu/xfer/gnuradio/volk/kernels/volk/asm/neon/volk_32fc_32f_dot_prod_32fc_a_neonasm.s
-- Adding source file: /home/ubuntu/xfer/gnuradio/volk/kernels/volk/asm/neon/volk_32fc_32f_dot_prod_32fc_a_neonasmvmla.s
-- Adding source file: /home/ubuntu/xfer/gnuradio/volk/kernels/volk/asm/neon/volk_32fc_32f_dot_prod_32fc_a_neonpipeline.s
-- Adding source file: /home/ubuntu/xfer/gnuradio/volk/kernels/volk/asm/neon/volk_32fc_32f_dot_prod_32fc_a_unrollasm.s
-- Adding source file: /home/ubuntu/xfer/gnuradio/volk/kernels/volk/asm/neon/volk_32fc_x2_dot_prod_32fc_a_neonasm.s
-- Adding source file: /home/ubuntu/xfer/gnuradio/volk/kernels/volk/asm/neon/volk_32fc_x2_dot_prod_32fc_a_neonasm_opttests.s
-- Adding source file: /home/ubuntu/xfer/gnuradio/volk/kernels/volk/asm/neon/volk_32fc_x2_multiply_32fc_a_neonasm.s
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- c flags:  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -Wall;
-- asm flags: -mfpu=neon -g
-- c flags:  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -Wall;
-- asm flags: -mfpu=neon -g
```
Is the machine "neon_orc" useful if "neonv7_hardfp_orc" is present? With GNU Radio 3.7.13.4, only "generic_orc" and "neon_hardfp_orc" are selected.

balister, 2018-12-21:
Watch only setting the machine etc flags in C++ flags, there may be C files that need them. Also have some pain with .s files.

drmpeg, 2018-12-21:
@balister Thanx for the tip. It makes a significant difference. New CMake invocation:

cmake -DCMAKE_INSTALL_PREFIX=/opt/gnuradio-3.7.12git -DCMAKE_CXX_FLAGS:STRING="-march=armv7-a -mfloat-abi=hard -mfpu=neon -mtune=cortex-a15" -DCMAKE_C_FLAGS:STRING="-march=armv7-a -mfloat-abi=hard -mfpu=neon -mtune=cortex-a15" ../

CMake output:

```
-- Performing Test neon_compile_result
-- Performing Test neon_compile_result - Success
-- Performing Test have_neonv7_result
-- Performing Test have_neonv7_result - Success
-- Performing Test have_neonv8_result
-- Performing Test have_neonv8_result - Failed
-- CPU is armv7, Overruled arch neonv8
-- Available architectures: generic;hardfp;neon;neonv7;orc;norc
-- Available machines: generic_orc;neon_orc;neonv7_hardfp_orc
-- BUILD TYPE = RELEASE
-- Base cflags = -O3 -DNDEBUG -march=armv7-a -mfloat-abi=hard -mfpu=neon -mtune=cortex-a15 -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -Wall
-- BUILD INFO ::: generic_orc ::: GNU ::: -O3 -DNDEBUG -march=armv7-a -mfloat-abi=hard -mfpu=neon -mtune=cortex-a15 -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -Wall 
-- BUILD INFO ::: neon_orc ::: GNU ::: -O3 -DNDEBUG -march=armv7-a -mfloat-abi=hard -mfpu=neon -mtune=cortex-a15 -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -Wall -funsafe-math-optimizations -mfpu=neon
-- BUILD INFO ::: neonv7_hardfp_orc ::: GNU ::: -O3 -DNDEBUG -march=armv7-a -mfloat-abi=hard -mfpu=neon -mtune=cortex-a15 -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -Wall -funsafe-math-optimizations -mfpu=neon -funsafe-math-optimizations -mfloat-abi=hard
-- Compiler Version: cc (Ubuntu/Linaro 7.3.0-27ubuntu1~18.04) 7.3.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
-- ---- Adding ASM files
-- -- Detected neon architecture; enabling ASM
-- Adding source file: /home/ubuntu/xfer/gnuradio/volk/kernels/volk/asm/neon/volk_16i_max_star_horizontal_16i.s
-- Adding source file: /home/ubuntu/xfer/gnuradio/volk/kernels/volk/asm/neon/volk_32f_s32f_multiply_32f_a_neonasm.s
-- Adding source file: /home/ubuntu/xfer/gnuradio/volk/kernels/volk/asm/neon/volk_32f_x2_add_32f_a_neonasm.s
-- Adding source file: /home/ubuntu/xfer/gnuradio/volk/kernels/volk/asm/neon/volk_32f_x2_add_32f_a_neonpipeline.s
-- Adding source file: /home/ubuntu/xfer/gnuradio/volk/kernels/volk/asm/neon/volk_32f_x2_dot_prod_32f_a_neonasm.s
-- Adding source file: /home/ubuntu/xfer/gnuradio/volk/kernels/volk/asm/neon/volk_32f_x2_dot_prod_32f_a_neonasm_opts.s
-- Adding source file: /home/ubuntu/xfer/gnuradio/volk/kernels/volk/asm/neon/volk_32fc_32f_dot_prod_32fc_a_neonasm.s
-- Adding source file: /home/ubuntu/xfer/gnuradio/volk/kernels/volk/asm/neon/volk_32fc_32f_dot_prod_32fc_a_neonasmvmla.s
-- Adding source file: /home/ubuntu/xfer/gnuradio/volk/kernels/volk/asm/neon/volk_32fc_32f_dot_prod_32fc_a_neonpipeline.s
-- Adding source file: /home/ubuntu/xfer/gnuradio/volk/kernels/volk/asm/neon/volk_32fc_32f_dot_prod_32fc_a_unrollasm.s
-- Adding source file: /home/ubuntu/xfer/gnuradio/volk/kernels/volk/asm/neon/volk_32fc_x2_dot_prod_32fc_a_neonasm.s
-- Adding source file: /home/ubuntu/xfer/gnuradio/volk/kernels/volk/asm/neon/volk_32fc_x2_dot_prod_32fc_a_neonasm_opttests.s
-- Adding source file: /home/ubuntu/xfer/gnuradio/volk/kernels/volk/asm/neon/volk_32fc_x2_multiply_32fc_a_neonasm.s
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- c flags: -march=armv7-a -mfloat-abi=hard -mfpu=neon -mtune=cortex-a15 -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -Wall;
-- asm flags: -mfpu=neon -g
-- c flags: -march=armv7-a -mfloat-abi=hard -mfpu=neon -mtune=cortex-a15 -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -Wall;
-- asm flags: -mfpu=neon -g
```

balister, 2018-12-21:
The OpenEmbedded SDK does this to get the platform specific flags into the build:

export CC="arm-oe-linux-gnueabi-gcc  -march=armv7-a -mfloat-abi=hard -mfpu=neon --sysroot=$SDKTARGETSYSROOT"
export CXX="arm-oe-linux-gnueabi-g++  -march=armv7-a -mfloat-abi=hard -mfpu=neon --sysroot=$SDKTARGETSYSROOT"
export CPP="arm-oe-linux-gnueabi-gcc -E  -march=armv7-a -mfloat-abi=hard -mfpu=neon --sysroot=$SDKTARGETSYSROOT"

noc0lour, 2019-03-08:
This is fixed in the latest volk master

closed 2019-03-08 by noc0lour