gnuradio/gnuradio #2201: tag propagation and stream_to_vector/vector_to_stream

adisuciu, 2018-11-26:
I think there is a bug with how tag propagation is handled in stream_to_vector / vector_to_stream blocks - It does not propagate in GNURadio 3.8

I attached a few screenshots from my machines (same flowgraph)

- one is from an older windows build (which works just fine) 
[image]

- the other is from GR 3.8 which only captures the first tag
[image]

noc0lour, 2018-11-26:
For the `3.8` build can you give me the commit-hash you built this off? We recently had several commits working on tag propagation using multi-precision arithmetic.

noc0lour, 2018-11-26:
Also tags on the `3.7` version look like they are on the wrong offset after conversion to and from the stream to vector / vector to stream operations. (But that might be due to "alignment" issues)

adisuciu, 2018-11-26:
The 3.8 is built off this - 8dc5fa49dc4c669c28ffb1216625ae92475c4ec9
The Windows build which works fine(ish) is 3.7.12

noc0lour, 2018-11-26:
Ok, strange. Testing this locally yields: default values for num_items on vector->stream (num_items: 2) works flawlessly. Every other value for num_items results in only the first tag arriving.

noc0lour, 2018-11-26:
Can you test with #2202?  This should fix the tag propagation.

PR #2202 (2018-11-27): runtime: Fix tag propagation
During the fix for long long on 32bit architectures a bug was introduced
which lead to offset being multiplied by (mp_rrate + one_half).

The correct solution is to multiply with mp_rrate and then add one_half.

Fixes #2201
commit 0733bc4f (2018-11-27): runtime: Fix tag propagation

During the fix for long long on 32bit architectures a bug was introduced
which lead to offset being multiplied by (mp_rrate + one_half).

The correct solution is to multiply with mp_rrate and then add one_half.

Fixes #2201
closed 2018-11-27 by noc0lour