gnuradio/gnuradio #2207: qa_python_message_passing flakiness

noc0lour, 2018-11-29:
From: https://ci.gnuradio.org/buildbot/#/builders/52/builds/250/steps/5/logs/stdio

204/351 Test #204: qa_python_message_passing ..........................***Failed    1.31 sec
F
======================================================================
FAIL: test_000 (__main__.test_python_message_passing)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/data/src/pull/2167/master/gr-blocks/python/blocks/qa_python_message_passing.py", line 115, in test_000
    self.assertEqual(num_msgs, len(msg_cons.msg_list))
AssertionError: 10 != 0
----------------------------------------------------------------------
Ran 1 test in 0.024s
FAILED (failures=1)

Pulled out of #1261 for better organization/triage of issues.

mbr0wn, 2019-01-04:
Still happening: https://ci.gnuradio.org/buildbot/#/builders/45/builds/458

noc0lour, 2019-01-05:
Maybe @osh has some suggestions in which way the qa test is using the message passing system in the wrong way?

argilo, 2022-04-07:
It would appear this was solved by #2564. I cannot reproduce it on the main branch, but I can if I revert #2564. I have also not seen any recent instances of this test failure in CI.

closed 2022-04-07 by argilo