gnuradio/gnuradio #2209: Compilation error in gr-zeromq

phansel, 2018-12-04:
As of commit 3359d67e02158199d24756155fb58ab43b7b66ef, I cannot build GNURadio.

`In file included from /home/paul/code/gnuradio/gr-zeromq/lib/pull_msg_source_impl.cc:30:0:
/home/paul/code/gnuradio/gr-zeromq/lib/pull_msg_source_impl.h: In member function ‘virtual std::__cxx11::string gr::zeromq::pull_msg_source_impl::last_endpoint()’:
/home/paul/code/gnuradio/gr-zeromq/lib/pull_msg_source_impl.h:55:30: error: ‘ZMQ_LAST_ENDPOINT’ was not declared in this scope
         d_socket->getsockopt(ZMQ_LAST_ENDPOINT, addr, &addr_len);
                              ^
[ 39%] Built target blocks_qa_gr_block.cc
[ 39%] Built target blocks_qa_gr_flowgraph.cc
gr-zeromq/lib/CMakeFiles/gnuradio-zeromq.dir/build.make:254: recipe for target 'gr-zeromq/lib/CMakeFiles/gnuradio-zeromq.dir/push_msg_sink_impl.cc.o' failed
make[2]: *** [gr-zeromq/lib/CMakeFiles/gnuradio-zeromq.dir/push_msg_sink_impl.cc.o] Error 1
`

Linux 4.15.0-39-generic #42~16.04.1-Ubuntu SMP Wed Oct 24 17:09:54 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

michaelld, 2018-12-04:
Do you have any `zmq` dev installed? If so, what version? Must be 3.2.0 or more recent for `ZMQ_LAST_ENDPOINT`.

michaelld, 2018-12-04:
I strongly doubt is was the commit mentioned that caused the issue, since it is generic C++ that has nothing to do with ZMQ. I'd believe it was this instead: < https://github.com/gnuradio/gnuradio/commit/b76a95c52f812181605248f1f590a199e2e5d0e3 >.

btashton, 2018-12-04:
Yikes, it looks like Ubuntu 16 is using 2.2

btashton, 2018-12-04:
Looks like you should be able to use the zeromq3 package. https://packages.ubuntu.com/source/xenial/zeromq3

phansel, 2018-12-12:
Thank you! I will update to zeromq3 and let you know if that fixes it.

phansel, 2018-12-15:
Yep, this worked.

closed 2019-02-25 by noc0lour