gnuradio/gnuradio #2214: QA: qa_pdu flakiness

noc0lour, 2018-12-05:
From https://ci.gnuradio.org/buildbot/#/builders/56/builds/337/steps/5/logs/stdio
```
198/360 Test #198: qa_pdu .............................................***Failed    0.59 sec
E..
======================================================================
ERROR: test_000 (__main__.test_pdu)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/data/src/pull/2211/master/gr-blocks/python/blocks/qa_pdu.py", line 77, in test_000
    result_msg = dbg.get_message(0)
  File "/cache/workdir/Fedora_29_0/build_Fedora_29/build/gr-blocks/python/blocks/../../swig/blocks_swig0.py", line 7074, in get_message
    return _blocks_swig0.message_debug_sptr_get_message(self, i)
RuntimeError: message_debug: index for message out of bounds.
----------------------------------------------------------------------
Ran 3 tests in 0.012s
FAILED (errors=1)
```

nickoe, 2019-03-13:
Maybe this can be closed? It looks like the qa_pdu test passes on all the buildbot tests consistently. Or are there still cases where this test fails?

noc0lour, 2019-03-13:
Hmmm, If we didn't do something about it it's probably not yet fixed.

noc0lour, 2019-03-13:
It's pretty rare to spot

mbr0wn, 2021-03-04:
Have not seen this in a very long time. Closing.

closed 2021-03-04 by mbr0wn