gnuradio/gnuradio #2215: qa: header_payload_demux flakiness

noc0lour, 2018-12-05:
From: https://ci.gnuradio.org/buildbot/#/builders/51/builds/331/steps/5/logs/stdio
```
321/360 Test #321: qa_header_payload_demux ............................***Failed    7.95 sec
F......
======================================================================
FAIL: test_001_headerpadding (__main__.qa_header_payload_demux)
Like test 1, but with header padding.
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/data/src/pull/2204/master/gr-digital/python/digital/qa_header_payload_demux.py", line 275, in test_001_headerpadding
    self.assertEqual(payload_sink.data(), payload)
AssertionError: Tuples differ: () != (5, 6, 7, 8, 9, 10, 11, 12, 13...
Second tuple contains 15 additional elements.
First extra element 0:
5
- ()
+ (5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19)
----------------------------------------------------------------------
Ran 7 tests in 3.883s
FAILED (failures=1)
Triggers at: 5 23 31
Random seed: 77172390
Triggers at: 5 23 31
Random seed: 77172390
```

mbr0wn, 2018-12-16:
I'm not able to reproduce this. I can see how the failure above would occur: The HPD waits for a message, which probably never came. In that case, the HDP never outputs anything into the payload output.

But why wouldn't it come? Hrm.

mbr0wn, 2018-12-16:
Maybe there was some issue with the message passing subsystem a while ago?

noc0lour, 2018-12-16:
That's not from a while ago. I've got a theory about the message passing having big time trouble in high load situations. Since the threads won't fire to the point where the flowgraph is shut down.

mbr0wn, 2018-12-16:
```
339/360 Test #339: qa_header_payload_demux ............................***Failed    6.15 sec
F......
======================================================================
FAIL: test_001_headerpadding (__main__.qa_header_payload_demux)
Like test 1, but with header padding.
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/cache/workdir/Ubuntu_16_04_64_remote_0/build_Ubuntu_16_04_64/src/gr-digital/python/digital/qa_header_payload_demux.py", line 274, in test_001_headerpadding
    self.assertEqual(header_sink.data(),  (0,) + header + (payload[0],))
AssertionError: Tuples differ: () != (0, 1, 2, 3, 5)
Second tuple contains 5 additional elements.
First extra element 0:
0
- ()
+ (0, 1, 2, 3, 5)
----------------------------------------------------------------------
Ran 7 tests in 2.783s
FAILED (failures=1)
Triggers at: 5 23 31
Random seed: 77249736
Triggers at: 5 23 31
Random seed: 77249736
```

Build: https://ci.gnuradio.org/buildbot/#/builders/49/builds/405/steps/7/logs/stdio

noc0lour, 2018-12-17:
We def. do not have "high" host contention right now.

mbr0wn, 2018-12-18:
You think this is a bug in the code itself? Either HPD, or HPD QA?

noc0lour, 2018-12-18:
Maybe, I actually didn't look into that yet.

PR #3115 (2020-01-24): qa: Fix the stop condition in qa_header_payload_demux
Tests nees to run until both payload_sink and header_sink recive the
specified amount of data or the time limit is reached and should not
stop when only one of the sinks has received enough data.

This should fix the sporadic tests failures.
Fixes #2215
closed 2020-01-24 by mbr0wn