gnuradio/gnuradio #2221: UHD: uhd sink not working

drmpeg, 2018-12-12:
The UHD sink seems pretty broken in 3.8. With an Ettus B210, I can't see any RF output with a spectrum analyzer even though the red TX LED does come on.

My guess is that it's not accepting the block parameters such as gain and/or frequency. Parameters that definitely don't work are Subdev Spec (can't select other channel with A:B) and Wire Format (selecting int16, int12 or int8 causes error).

mbr0wn, 2018-12-15:
Reopening since there are still issues. To quote @drmpeg:

> I was able to transmit RF at the correct frequency, so big improvement. But still some issues.
> 
>     Subdev Spec still doesn't work. Can't select other channel with A:B.
> 
>     Wire Format still doesn't work. Can't select anything except Automatic.
> 
>     Sync parameter doesn't "stick". Always returns to "Unknown PPS" when reloading flow graph.
> 
>     Bandwidth value of 0 doesn't work as before. Error message:
> 
> WARNING] [AD936X] The requested bandwidth 0.000000 MHz is out of range (0.200000 - 56.000000 MHz).
> The bandwidth has been forced to 0.200000 MHz.

mbr0wn, 2018-12-16:
OK I see -- 3.7 would ignore the BW setting when the value was zer.

drmpeg, 2018-12-16:
So then what is the actual bandwidth being set to?

mbr0wn, 2018-12-16:
UHD will receive a request for a 0 Hz analog bandwidth, and will then coerce it to the smallest possible value. I'll change the behaviour to the 3.7 behaviour in gr-uhd.

mbr0wn, 2018-12-16:
...in which case, 0 bandwidth means "ignore this value".

drmpeg, 2018-12-16:
I meant if you set bandwidth=0 in 3.7, what is the actual hardware analog bandwidth being set to. I wrote a little Python script to find out, and it turns out to be 56 MHz.

However, there must be another filter somewhere that gets rid of the alias, since the signal looks fine.

Now that I know what's actually going on, I'd have to say that the new behavior is correct. That is, setting bandwidth=0 should get coerced to 0.2 MHz.

Sorry for the confusion.

mbr0wn, 2018-12-16:
@drmpeg keeps finding stuff :smile: 

Here's what he writes:

Looks good. However, still a couple of issues remaining.

1) Show Command Port option doesn't shut of command port display.

2) Absolute Gain takes an initial gain between 0 and 89, but when trying to change gain during runtime, the following error message occurs.
```
Traceback (most recent call last):
  File "/opt/gnuradio-3.7.12git/lib/python3.6/dist-packages/gnuradio/qtgui/range.py", line 197, in changed
    self.notifyChanged(self.rangeType(value))
  File "/opt/gnuradio-3.7.12git/lib/python3.6/dist-packages/gnuradio/qtgui/range.py", line 243, in counterChanged
    self.notifyChanged(self.rangeType(value))
  File "/home/re/xfer/top_block.py", line 217, in set_tx_gain
    self.uhd_usrp_sink_0.set_normalized_gain(self.tx_gain, 0)
  File "/opt/gnuradio-3.7.12git/lib/python3.6/dist-packages/gnuradio/uhd/uhd_swig.py", line 4692, in set_normalized_gain
    return _uhd_swig.usrp_sink_sptr_set_normalized_gain(self, norm_gain, chan)
RuntimeError: RuntimeError: Normalized gain out of range, must be in [0, 1].
```

mbr0wn, 2018-12-16:
https://github.com/gnuradio/gnuradio/pull/2241

PR #2227 (2018-12-15): uhd: Fix GRC bindings (gain, freq...)
The YML file generated during CMake was simply not correct. Frequency,
gain, etc. weren't propagated properly to the underlying C++ object.

Fixes #2221.
PR #2231 (2018-12-16): uhd: More fixes to GRC bindings (time/clock source, etc.)
- Subdev spec wasn't being applied at all
- time/clock source, neither
- Sync wouldn't stay the same value when it was set to "don't sync"
- Bandwidth is no longer set when the value is 0, allowing for a default
  behaviour
- OTW format wasn't being applied properly

Fixes #2221.
PR #2241 (2018-12-18): uhd: Fix even more GRC binding issues
- The callback for gain would always use normalized gain, which would
  cause errors since the non-normalized gain range would cause
  exceptions
- The "Show Command Port" menu was removed, it didn't work, but no other
  block has this.

Finally fixes #2221.
closed 2018-12-18 by mbr0wn