gnuradio/gnuradio #2222: Consolidate Manuals (Sphinx, Doxygen)

mbr0wn, 2018-12-13:
See also  #2118. We should change the title once we know which way to consolidate.

777arc, 2018-12-16:
We should also think about what people are actually using the Python manual to reference.  A majority of it is just the python names of all the blocks, and we don't need sphinx or doxygen to be able to provide that kind of list.  The Runtime category has some important stuff, although right now it's not split up in terms of whether you would call the function from within a block or not, and that feels confusing to me. 
 There's only a handful of helper classes, they can all be lumped into one category.  In general it seems like the source code is a better reference than this Sphinx manual.  

As far as how to move forward...  First off, I love Sphinx and RST in general.  But my thought is to change as little as possible (thus avoid migrating the C++ API to Sphinx), and just remove the Sphinx manual, moving the useful information into the Doxygen manual, or possibly the wiki if it's not specific to a block/class/function/etc.  Problem is, no one seems to actually use the Sphinx manual.  We should try removing it for a week or a month and replace it with an HTML form that lets people type in what they were intending to reference.  

Either way, we need to reduce the number of locations GNU Radio documentation exists within, if we ever want to get it to be in an organized state.

jdemel, 2018-12-19:
A quick search on how to use Doxygen for Python or Sphinx for C++ looks like Doxygen for Python might be the way to go. Though, I'm pretty sure we want to use Python docstring and thus, we will not be able to use special Doxygen syntax in Python. Though, this might be advantageous  because it does not screw with the Python docstring system.

777arc, 2019-01-17:
Can we agree on removing Sphinx, so I can feel comfortable starting work on getting the useful Python docs incorporated into our Doxygen manual?

mbr0wn, 2019-01-19:
Let's kill it! We just need to be sure how we want to yank docstrings into doxygen.

777arc, 2019-03-13:
@noc0lour can we try temporarily (e.g. 1 month) replacing the Sphinx manual site with a page that warns users the manual is going to be removed and email this address (it can be my email) if you find the Sphinx manual useful, and include what specifically you use it to look up.  I can create the HTML of that page, but is there an easy way to make all GETS to gnuradio.org/doc/sphinx/* be directed to this page?  I think it's a great precursor to actually removing it, which we all want to do.  Maybe we will find out something new about what people use it for.

bhilburn, 2019-06-20:
We chatted about this in the leads call. We will be removing the Sphinx manual and consolidating.

777arc, 2019-06-20:
I'll try to create that landing page that I described above, and see if anyone comments about it.

777arc, 2019-06-22:
See #2569

mbr0wn, 2021-05-14:
We've removed the sphinx manual. Closing.

closed 2021-05-14 by mbr0wn