gnuradio/gnuradio #2223: Building with -DENABLE_STATIC_LIBS fails

haakov, 2018-12-13:
CMake output:
```
$ cmake .. -DENABLE_STATIC_LIBS=ON
(...)
-- Configuring gr-ctrlport support...
--   Dependency Boost_FOUND = 1
--   Dependency SWIG_FOUND = TRUE
--   Dependency SWIG_VERSION_CHECK = TRUE
--   Dependency ENABLE_GNURADIO_RUNTIME = ON
--   Enabling gr-ctrlport support.
--   Override with -DENABLE_GR_CTRLPORT=ON/OFF
-- Loading build date Thu, 13 Dec 2018 21:23:20 into constants...
-- Loading version 3.8tech-preview-156-g8628adf6 into constants...
CMake Error at gnuradio-runtime/lib/pmt/CMakeLists.txt:62 (add_dependencies):
  add_dependencies called with incorrect number of arguments

-- Checking for module 'thrift'
--   Package 'thrift', required by 'virtual:world', not found
-- Binary 'thrift' not found.
-- TRY_SHM_VMCIRCBUF set to ON.
-- 
-- Configuring gr-blocks support...
--   Dependency Boost_FOUND = 1
--   Dependency ENABLE_VOLK = ON
--   Dependency ENABLE_GNURADIO_RUNTIME = ON
--   Enabling gr-blocks support.
--   Override with -DENABLE_GR_BLOCKS=ON/OFF
CMake Error at gr-blocks/lib/CMakeLists.txt:263 (add_dependencies):
  add_dependencies called with incorrect number of arguments
(...)
```

Looks like the `gnuradio-blocks_static` variable is empty: 

https://github.com/gnuradio/gnuradio/blob/8628adf64d14c9d3d24e28dc4f314337d705ca63/gr-blocks/lib/CMakeLists.txt#L261-L263

nickoe, 2020-03-26:
@haakov Is this still an issue. I just tried to test this on master of 414a495d639ffb817dd3c04d771719e938f50daf, and it makes the gnuradio-blocks target just fine.

haakov, 2020-04-03:
I haven't run it in a while, so I'll do that tomorrow and get back to you :)

haakov, 2020-04-04:
yeah gnuradio-blocks works for me too. I'm also running into a problem with volk, but it might be my own fault

nickoe, 2020-04-05:
@haakov As this issue is quite old, I will take the liberty to close it. I am not sure how to verify that building the static libs are working as _intended_, but lets make that another issue if that is the case.

Yeah, the volk issue was probably also self made for me and resolved somehow.

closed 2020-04-05 by nickoe