gnuradio/gnuradio #2229: digital: OFDM Cyclic Prefixer block Length Tag Key value doesn't "stick"

drmpeg, 2018-12-15:
For DTV OFDM based flow graphs, the Length Tag Key is set to blank. When the flow graph is reloaded, the Length Tag Key always reverts to the default "frame_len".

drmpeg, 2018-12-17:
This can be solved by using empty quotes ("") instead of blank.

closed 2018-12-17 by drmpeg

noc0lour, 2018-12-18:
Hm, so doesn't need a fix? Or should it do the same thing for blank as well?

drmpeg, 2018-12-18:
I guess blank would be a nice feature. I'm regenerating all of the DTV example flow graphs, so I can live with empty quotes (which is why I closed the issue).

noc0lour, 2018-12-18:
Ok, I see. I'm not too fit in the current yaml format, maybe @haakov might be able to tell if it would be possible to implement such feature (or at least think about it and open an issue for that).

haakov, 2018-12-19:
Interesting! Actually, GRC discards all blank parameters upon creating the `.grc` file. I've created a PR for it: https://github.com/gnuradio/gnuradio/pull/2254