gnuradio/gnuradio #2239: Clarify: Can we use Python to generate .yml files at config time?

mbr0wn, 2018-12-16:
See comments here: 

- https://github.com/gnuradio/gnuradio/pull/2196
- https://github.com/gnuradio/gnuradio/pull/2234

The USRP blocks use this, and the variable struct. The qtgui time sink doesn't, but it still has a Python script.

noc0lour, 2018-12-22:
If possible I'd like to avoid generating yaml files during a build without explicit tests afterwards (Having verification tests for yaml files would be great either way, but I'd say for python generated one it's a must).

haakov, 2018-12-31:
We can start using YAML's anchors and extensions, which I am fairly certain will enable us to rewrite the `variable_struct` Python file from:
https://github.com/gnuradio/gnuradio/blob/529531fa7dc4ad7c74393dcdd07870d347f34b2e/grc/blocks/variable_struct.block.yml.py#L51-L55

to something like this, given that we define an anchor called `field`:

```
-  <<: *field
   { id: 'field0', label: 'Field 0' }
```

In other words, we can reduce the line count in the resulting YAML file, _but_ it will still have to be generated somehow.

One alternative is to utilize a templating library, but I suppose we might not want to bring in more dependencies. (like Jinja2, in Ansible's case)

marcusmueller, 2023-12-22:
seeing  gr-uhd and gr-qtgui, I'd say this has been answered by the fabulously normative power of the factual.

closed 2023-12-22 by marcusmueller