gnuradio/gnuradio #2259: Commit 7daef13 completely breaks GRC

drmpeg, 2018-12-20:
https://github.com/gnuradio/gnuradio/commit/7daef13332e7ae8fbe7bf90d1a80240c315c7e85 completely and utterly breaks GNU Radio Companion. Almost all blocks are missing.

[image]

noc0lour, 2018-12-20:
Oh wow, well, need to figure out what happened there.

noc0lour, 2018-12-22:
Hm, I rebuilt with 018658c786b1c178e3fa0113c761641c81a83c0a and everything seems to work? There was a change in the CMake, maybe you need to wipe the buildir & recreate the build files?

drmpeg, 2018-12-22:
I'll give it another try.

drmpeg, 2018-12-23:
Still failing here. Ubuntu 18.10 with CMake 3.12.1.

Also fails on my ARM system. Ubuntu 18.04.1 with CMake 3.10.2.

drmpeg, 2018-12-23:
On my system, the variable ENABLE_GRC doesn't get set until all the other components are configured (that is, the gnuradio-companion component is configured last). Here's my CMake output. I've put the status message "GRC enabled here!!!" just after ENABLE_GRC gets set.
```
-- The CXX compiler identification is GNU 8.2.0
-- The C compiler identification is GNU 8.2.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type not specified: defaulting to release.
-- Build type set to Release.
-- Found Git: /usr/bin/git  
-- Extracting version information from git describe...
-- Performing Test HAVE_VISIBILITY_HIDDEN
-- Performing Test HAVE_VISIBILITY_HIDDEN - Success
-- Performing Test HAVE_WARN_SIGN_COMPARE
-- Performing Test HAVE_WARN_SIGN_COMPARE - Success
-- Performing Test HAVE_WARN_ALL
-- Performing Test HAVE_WARN_ALL - Success
-- Performing Test HAVE_WARN_NO_UNINITIALIZED
-- Performing Test HAVE_WARN_NO_UNINITIALIZED - Success
-- Compiler Version: cc (Ubuntu 8.2.0-7ubuntu1) 8.2.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
-- Compiler Flags: /usr/bin/cc:::-O3 -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
/usr/bin/c++:::-O3 -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
-- ADDING PERF COUNTERS
-- Building Static Libraries: OFF
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Boost version: 1.67.0
-- Found the following Boost libraries:
--   date_time
--   program_options
--   filesystem
--   system
--   regex
--   thread
--   unit_test_framework
--   chrono
--   atomic
-- PYTHON_EXECUTABLE not set - using default python3
-- Use -DPYTHON_EXECUTABLE=/path/to/python2 to build for python2.
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.7", minimum required is "3.4") 
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.6m.so (found suitable exact version "3.6.7") 
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.6m.so (found suitable version "3.6.7", minimum required is "2.7") 
-- Python checking for six - python 2 and 3 compatibility library - found
-- 
-- Checking for module SWIG
-- Found SWIG version 3.0.12.
-- Found SWIG: /usr/bin/swig3.0  
-- 
-- The build system will automatically enable all components.
-- Use -DENABLE_DEFAULT=OFF to disable components by default.
-- 
[... 478 lines cut ...]
--   * * oss
--   * gr-channels
--   * gr-qtgui
--   * gr-trellis
--   * gr-uhd
--   * gr-utils
--   * gr-video-sdl
--   * gr-vocoder
--   * gr-wavelet
--   * gr-zeromq
--   * gnuradio-companion
-- 
-- ######################################################
-- # Gnuradio disabled components                        
-- ######################################################
--   * sphinx
--   * gr-comedi
-- 
-- Using install prefix: /opt/gnuradio-3.7.12git
-- Building for version: 3.8tech-preview-191-g018658c7 / 3.8git
-- Configuring done
-- Generating done
-- Build files have been written to: /home/re/xfer/gnuradiotest/build
```
Perhaps you didn't delete you install prefix directory?

bangyanh, 2018-12-25:
hi, I have also met this problem. well it seems that all the compiling and building are normal

drmpeg, 2018-12-25:
You can work around this issue by setting ENABLE_GRC manually.
```
cmake -DENABLE_GRC=ON ../
```

noc0lour, 2018-12-25:
@drmpeg you're right. The GRC subdir is enabled as last directory. I don't know if there is reasoning to do so. I don't really see a real build dependency on other GNU Radio components. So possibly it could be put in front.

mbr0wn, 2018-12-29:
Ah I see. I didn't delete *my* build directory before testing.

PR #2272 (2018-12-30): fixup! cmake: Only go into grc/ subdirs when ENABLE_GRC=ON
The grc subdir was parsed last, making ENABLE_GRC appear false in all
components, thus disabling blocks.

Fixes #2259.
closed 2018-12-30 by noc0lour