gnuradio/gnuradio #2260: GRC: Polar Encoder Definition block is broken

jdemel, 2018-12-20:
If you try to run e.g. `gr-fec/examples/polar_code_example.grc` it won't work because `POLAR Encoder Definition` is broken. The issue comes up with the blocks `Packed Bits` attribute. GRC is not able to run `eval` against this block. The attribute is not recognized as a boolean. Though, the Python interface to instantiate this block seems to work fine.

haakov, 2018-12-25:
https://github.com/gnuradio/gnuradio/pull/2265

Does this PR fix the problem?

jdemel, 2018-12-31:
This did fix the issue. Thanks. Also, it lead me in the right direction to fix some more issues. This issue can be closed. @marcusmueller

jdemel, 2019-01-05:
@marcusmueller I assume this means yes. The issue can be closed.

closed 2019-01-05 by jdemel