gnuradio/gnuradio #2262: The GR DTV blocks SSE2 tests succeeds also when building on ARM

ast, 2018-12-25:
Tested on Raspberry Pi with clang 3.8.

ast, 2018-12-25:
Should it be flagged as a bug?

balister, 2018-12-25:
So the problem is the sse2 test succeeds when run with an arm compiler?

ast, 2018-12-25:
Same things happen in volk but then they filter it out because arch is not x86. I think default clang is multiarch and compiles it just fine... Not sure, have to test more.

balister, 2018-12-25:
Sounds like a clang bug :) Or you are running clang that is producing x86 output. Define CC to have the right arch flags. Something like this:
export CC="arm-oe-linux-gnueabi-gcc  -march=armv7-a -mfloat-abi=hard -mfpu=neon --sysroot=$SDKTARGETSYSROOT"
export CXX="arm-oe-linux-gnueabi-g++  -march=armv7-a -mfloat-abi=hard -mfpu=neon --sysroot=$SDKTARGETSYSROOT"
export CPP="arm-oe-linux-gnueabi-gcc -E  -march=armv7-a -mfloat-abi=hard -mfpu=neon --sysroot=$SDKTARGETSYSROOT"

drmpeg, 2018-12-25:
@balister Passing the correct flags doesn't help. Already tried that.

drmpeg, 2018-12-25:
BTW, I'm using clang-6.0 on arm.

balister, 2018-12-25:
This looks bad:
https://github.com/gnuradio/gnuradio/blob/master/gr-dtv/lib/CMakeLists.txt#L120
I'd replace that with a try compile to see if the compiler can build sse2 code. Using flags to check for the presence of simd is not great, use try_compile blocks.

drmpeg, 2018-12-26:
@balister Agreed, although that's been in there since 10/2015. I guess there's not too many clang users on arm.

drmpeg, 2018-12-26:
Pull request submitted. See https://github.com/gnuradio/gnuradio/pull/2266

ast, 2018-12-26:
There is a very similar bug in libvolk I submitted, not sure if you are also involved in that?
https://github.com/gnuradio/volk/issues/221

Thanks for the quick fix!

closed 2019-01-10 by mbr0wn