gnuradio/gnuradio #2264: Test HAVE_SHM_OPEN requires -pthread before -lrt when building for raspberry pi

ast, 2018-12-25:
gnuradio-runtime/ConfigChecks.cmake line 162

https://github.com/gnuradio/gnuradio/blob/8b353303ab338abd302af088582d5c47bd1ccdf0/gnuradio-runtime/ConfigChecks.cmake#L134

otherwise:

```undefined reference to `__shm_directory'```

ast, 2019-04-15:
This is still the case with the current head and clang-8.

It works if I add `-pthread`.

sab24, 2019-06-08:
Hi,

I tried compiling HEAD on armv7 archlinux distro. I use cmake 3.14.5 and it automatically adds -lpthread. What distro did you use? And what CMake version?

```
-- Binary 'thrift' not found.
-- Looking for C++ include windows.h
CMake Warning (dev) at /usr/share/cmake-3.14/Modules/CheckIncludeFileCXX.cmake:79 (message):
  Policy CMP0075 is not set: Include file check macros honor
  CMAKE_REQUIRED_LIBRARIES.  Run "cmake --help-policy CMP0075" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.

  CMAKE_REQUIRED_LIBRARIES is set to:

    -lpthread

  For compatibility with CMake 3.11 and below this check is ignoring it.
Call Stack (most recent call first):
  gnuradio-runtime/lib/CMakeLists.txt:230 (check_include_file_cxx)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Looking for C++ include windows.h - not found
-- TRY_SHM_VMCIRCBUF set to ON.
--
-- Python checking for python2 >= 2.7.6 or python3 >= 3.4.0 - found
-- Python checking for PyYAML >= 3.10 - found
-- Python checking for mako >= 0.9.1 - found
-- Python checking for pygobject >= 2.28.6 - found
-- Python checking for Gtk (GI) >= 3.10.8 - found
-- Python checking for Cairo (GI) >= 1.0 - found
-- Python checking for PangoCairo (GI) >= 1.0 - found
-- Python checking for numpy - found ```

bhilburn, 2019-06-20:
@ast - Can you file a PR for this? We'd like to get this in to `master` for 3.8 ASAP

ast, 2019-06-20:
This was with clang 7.0, there might have been some cmake weirdness in flag detection. Like @sab24 says it was added to `CMAKE_REQUIRED_LIBRARIES` for him.

It's probably better to investigate it properly or it might mask some other problem. What do you think?

marcusmueller, 2023-12-22:
I think seeing all major RPi-targetting distros have managed to build, so I think this is very very likely done

closed 2023-12-22 by marcusmueller