gnuradio/gnuradio #2267: gr-atsc fails with clang 7

ast, 2018-12-26:
Maybe more of a build file bug?

```make[2]: *** No rule to make target '../gr-atsc/lib/atsci_viterbi_gen', needed by 'gr-atsc/lib/atsci_viterbi_mux.cc'.  Stop.
CMakeFiles/Makefile2:1878: recipe for target 'gr-atsc/lib/CMakeFiles/gnuradio-atsc.dir/all' failed
make[1]: *** [gr-atsc/lib/CMakeFiles/gnuradio-atsc.dir/all] Error 2
Makefile:162: recipe for target 'all' failed
make: *** [all] Error 2```

```cmake version 3.10.2```
```clang version 7.0.0 (tags/RELEASE_700/final)```

drmpeg, 2018-12-27:
There's nothing useful in gr-atsc. All of the functionality has been replaced or enhanced in gr-dtv. In 3.8, gr-atsc has been deleted.

ast, 2018-12-27:
Good.

Sorry about that. Should have checked better upstream. Just wanted start experimentatiok with a stable release. Will move on to upstream now.

drmpeg, 2018-12-27:
Also, I can't reproduce this. maint-3.7 builds okay with clang-6.0 on BeagleBoard-X15.

ast, 2018-12-27:
Ok even better. Maybe I messed something up or maybe it's a raspbian bug. Do you use any special flags or just ```cmake ..```? I will try again.

drmpeg, 2018-12-27:
Here's my CMake. It doesn't configure volk very well, but you know all about that.
```
cmake -DCMAKE_INSTALL_PREFIX=/opt/gnuradio-3.7.12git -DCMAKE_CXX_FLAGS:STRING="-march=armv7-a -mfloat-abi=hard -mfpu=neon -mtune=cortex-a15 -Werror=unused-command-line-argument" -DCMAKE_C_FLAGS:STRING="-march=armv7-a -mfloat-abi=hard -mfpu=neon -mtune=cortex-a15 -Werror=unused-command-line-argument" ../
```

closed 2019-02-25 by noc0lour