gnuradio/gnuradio #2268: gnuradio-runtime.pc missing boost dependencies

ast, 2018-12-26:
Depends on boost but they are not in ```Libs``` so if you build a program that uses the C++ API you need to list boost as a dependency, using pkg-config is not enough. 

Adding ```-llibboost_system```should be enough.

```
prefix=/usr/local
exec_prefix=${prefix}
libdir=${exec_prefix}/lib
includedir=${prefix}/include

Name: gnuradio-runtime
Description: GNU Radio core runtime infrastructure
Requires:
Version: 3.7.13.4
Libs: -L${libdir} -lgnuradio-runtime -lgnuradio-pmt -llog4cpp
Cflags: -I${includedir}
```

marcusmueller, 2019-04-21:
It feels like we should be able to solve this rather elegantly with the modern CMake that Mait and @noc0lour encouraged, but I don't know how.

michaelld, 2019-04-24:
At least on OSX, the DYLD needs -all- library dependencies ... in modern CMake notation: Volk::volk, Boost::date_time, Boost::program_options, Boost::filesystem, Boost::system, Boost::regex, Boost::thread, Boost::unit_test_framework, Boost::chrono, Boost::atomic.

Modern CMake targets should be usable via some PROPERTY retrieval & substitution via CMake ... might have to create temporary variables for the substitution; not sure if CMake allows target PROPERTY substitutions directly.

I do agree that this should be fixed, even if the vast majority of projects needing GR use CMake & thus don't use PKGCONFIG or this file.

bhilburn, 2019-06-20:
We suspect there are changes already made, here, tagging @noc0lour to look at this issue further.

marcusmueller, 2019-07-15:
ping @noc0lour

noc0lour, 2019-07-17:
@ast What's your usecase for using the pkg-config ? With the new cmake in place there is little to no need for having pkg-config out there.

noc0lour, 2019-07-17:
For generating the right pkgconfig this is probably a good starting point
https://stackoverflow.com/questions/44292462/how-to-auto-generate-pkgconfig-files-from-cmake-targets

ast, 2019-07-17:
@noc0lour use case is building an application using the C++ API.

noc0lour, 2019-07-17:
So this still needs to be done for the packageconfig files, manually updating packagconfig files is not really manageable.

NeuroForLunch, 2021-06-24:
I assume this issue has been resolved?

marcusmueller, 2023-12-22:
Not really solved, we seem to be missing spdlog dependencies now