gnuradio/gnuradio #2270: skiphead: heap-use-after-free detected by using ASAN

noc0lour, 2018-12-29:
```
10/360 Test #210: qa_skiphead ........................................***Failed    2.12 sec
......=================================================================
==13820==ERROR: AddressSanitizer: heap-use-after-free on address 0x60600000abe8 at pc 0x7f97f34907a1 bp 0x7fff76018f00 sp 0x7fff76018ef8
READ of size 8 at 0x60600000abe8 thread T0
    #0 0x7f97f34907a0 in boost::intrusive_ptr<pmt::pmt_base>::get() const /usr/include/boost/smart_ptr/intrusive_ptr.hpp:181:16
    #1 0x7f97f34c23c8 in bool boost::operator==<pmt::pmt_base, pmt::pmt_base>(boost::intrusive_ptr<pmt::pmt_base> const&, boost::intrusive_ptr<pmt::pmt_base> const&) /usr/include/boost/smart_ptr/intrusive_ptr.hpp:220:14
    #2 0x7f97f34ac1a6 in pmt::is_null(boost::intrusive_ptr<pmt::pmt_base> const&) /home/andrej/git/gnuradio/gnuradio/gnuradio-runtime/lib/pmt/pmt.cc:501:12
    #3 0x7f97f08a4d5a in _wrap_is_null(_object*, _object*, _object*) /home/andrej/git/gnuradio/gnuradio/build/gnuradio-runtime/swig/pmt_swigPYTHON_wrap.cxx:32965:22
    #4 0x7f97f9411cc3 in _PyCFunction_FastCallDict (/usr/lib64/libpython3.6m.so.1.0+0xb7cc3)
    #5 0x7f97f94860d4  (/usr/lib64/libpython3.6m.so.1.0+0x12c0d4)
    #6 0x7f97f9489de9 in _PyEval_EvalFrameDefault (/usr/lib64/libpython3.6m.so.1.0+0x12fde9)
    #7 0x7f97f948520f  (/usr/lib64/libpython3.6m.so.1.0+0x12b20f)
    #8 0x7f97f9485fec  (/usr/lib64/libpython3.6m.so.1.0+0x12bfec)
    #9 0x7f97f9489de9 in _PyEval_EvalFrameDefault (/usr/lib64/libpython3.6m.so.1.0+0x12fde9)
    #10 0x7f97f948520f  (/usr/lib64/libpython3.6m.so.1.0+0x12b20f)
    #11 0x7f97f9485fec  (/usr/lib64/libpython3.6m.so.1.0+0x12bfec)
    #12 0x7f97f9489de9 in _PyEval_EvalFrameDefault (/usr/lib64/libpython3.6m.so.1.0+0x12fde9)
    #13 0x7f97f948520f  (/usr/lib64/libpython3.6m.so.1.0+0x12b20f)
    #14 0x7f97f9485fec  (/usr/lib64/libpython3.6m.so.1.0+0x12bfec)
    #15 0x7f97f9489de9 in _PyEval_EvalFrameDefault (/usr/lib64/libpython3.6m.so.1.0+0x12fde9)
    #16 0x7f97f9485c3b  (/usr/lib64/libpython3.6m.so.1.0+0x12bc3b)
    #17 0x7f97f948e03b in _PyFunction_FastCallDict (/usr/lib64/libpython3.6m.so.1.0+0x13403b)
    #18 0x7f97f93c709d in _PyObject_FastCallDict (/usr/lib64/libpython3.6m.so.1.0+0x6d09d)
    #19 0x7f97f93c72bb in _PyObject_Call_Prepend (/usr/lib64/libpython3.6m.so.1.0+0x6d2bb)
    #20 0x7f97f93c6e46 in PyObject_Call (/usr/lib64/libpython3.6m.so.1.0+0x6ce46)
    #21 0x7f97f94876ce in _PyEval_EvalFrameDefault (/usr/lib64/libpython3.6m.so.1.0+0x12d6ce)
    #22 0x7f97f9485c3b  (/usr/lib64/libpython3.6m.so.1.0+0x12bc3b)
    #23 0x7f97f948e0dc in _PyFunction_FastCallDict (/usr/lib64/libpython3.6m.so.1.0+0x1340dc)
    #24 0x7f97f93c709d in _PyObject_FastCallDict (/usr/lib64/libpython3.6m.so.1.0+0x6d09d)
    #25 0x7f97f93c72bb in _PyObject_Call_Prepend (/usr/lib64/libpython3.6m.so.1.0+0x6d2bb)
    #26 0x7f97f93c6e46 in PyObject_Call (/usr/lib64/libpython3.6m.so.1.0+0x6ce46)
    #27 0x7f97f94299cf  (/usr/lib64/libpython3.6m.so.1.0+0xcf9cf)
    #28 0x7f97f93c6fb8 in _PyObject_FastCallDict (/usr/lib64/libpython3.6m.so.1.0+0x6cfb8)
    #29 0x7f97f9485f32  (/usr/lib64/libpython3.6m.so.1.0+0x12bf32)
    #30 0x7f97f9489de9 in _PyEval_EvalFrameDefault (/usr/lib64/libpython3.6m.so.1.0+0x12fde9)
    #31 0x7f97f9485c3b  (/usr/lib64/libpython3.6m.so.1.0+0x12bc3b)
    #32 0x7f97f948e03b in _PyFunction_FastCallDict (/usr/lib64/libpython3.6m.so.1.0+0x13403b)
    #33 0x7f97f93c709d in _PyObject_FastCallDict (/usr/lib64/libpython3.6m.so.1.0+0x6d09d)
    #34 0x7f97f93c72bb in _PyObject_Call_Prepend (/usr/lib64/libpython3.6m.so.1.0+0x6d2bb)
    #35 0x7f97f93c6e46 in PyObject_Call (/usr/lib64/libpython3.6m.so.1.0+0x6ce46)
    #36 0x7f97f94876ce in _PyEval_EvalFrameDefault (/usr/lib64/libpython3.6m.so.1.0+0x12d6ce)
    #37 0x7f97f9485c3b  (/usr/lib64/libpython3.6m.so.1.0+0x12bc3b)
    #38 0x7f97f948e0dc in _PyFunction_FastCallDict (/usr/lib64/libpython3.6m.so.1.0+0x1340dc)
    #39 0x7f97f93c709d in _PyObject_FastCallDict (/usr/lib64/libpython3.6m.so.1.0+0x6d09d)
    #40 0x7f97f93c72bb in _PyObject_Call_Prepend (/usr/lib64/libpython3.6m.so.1.0+0x6d2bb)
    #41 0x7f97f93c6e46 in PyObject_Call (/usr/lib64/libpython3.6m.so.1.0+0x6ce46)
    #42 0x7f97f94299cf  (/usr/lib64/libpython3.6m.so.1.0+0xcf9cf)
    #43 0x7f97f93c6fb8 in _PyObject_FastCallDict (/usr/lib64/libpython3.6m.so.1.0+0x6cfb8)
    #44 0x7f97f9485f32  (/usr/lib64/libpython3.6m.so.1.0+0x12bf32)
    #45 0x7f97f9489de9 in _PyEval_EvalFrameDefault (/usr/lib64/libpython3.6m.so.1.0+0x12fde9)
    #46 0x7f97f9485c3b  (/usr/lib64/libpython3.6m.so.1.0+0x12bc3b)
    #47 0x7f97f948e03b in _PyFunction_FastCallDict (/usr/lib64/libpython3.6m.so.1.0+0x13403b)
    #48 0x7f97f93c709d in _PyObject_FastCallDict (/usr/lib64/libpython3.6m.so.1.0+0x6d09d)
    #49 0x7f97f93c72bb in _PyObject_Call_Prepend (/usr/lib64/libpython3.6m.so.1.0+0x6d2bb)
    #50 0x7f97f93c6e46 in PyObject_Call (/usr/lib64/libpython3.6m.so.1.0+0x6ce46)
    #51 0x7f97f94876ce in _PyEval_EvalFrameDefault (/usr/lib64/libpython3.6m.so.1.0+0x12d6ce)
    #52 0x7f97f9485c3b  (/usr/lib64/libpython3.6m.so.1.0+0x12bc3b)
    #53 0x7f97f948e0dc in _PyFunction_FastCallDict (/usr/lib64/libpython3.6m.so.1.0+0x1340dc)
    #54 0x7f97f93c709d in _PyObject_FastCallDict (/usr/lib64/libpython3.6m.so.1.0+0x6d09d)
    #55 0x7f97f93c72bb in _PyObject_Call_Prepend (/usr/lib64/libpython3.6m.so.1.0+0x6d2bb)
    #56 0x7f97f93c6e46 in PyObject_Call (/usr/lib64/libpython3.6m.so.1.0+0x6ce46)
    #57 0x7f97f94299cf  (/usr/lib64/libpython3.6m.so.1.0+0xcf9cf)
    #58 0x7f97f93c6fb8 in _PyObject_FastCallDict (/usr/lib64/libpython3.6m.so.1.0+0x6cfb8)
    #59 0x7f97f9485f32  (/usr/lib64/libpython3.6m.so.1.0+0x12bf32)
    #60 0x7f97f9489de9 in _PyEval_EvalFrameDefault (/usr/lib64/libpython3.6m.so.1.0+0x12fde9)
    #61 0x7f97f948520f  (/usr/lib64/libpython3.6m.so.1.0+0x12b20f)
    #62 0x7f97f9485fec  (/usr/lib64/libpython3.6m.so.1.0+0x12bfec)
    #63 0x7f97f9489de9 in _PyEval_EvalFrameDefault (/usr/lib64/libpython3.6m.so.1.0+0x12fde9)
    #64 0x7f97f948520f  (/usr/lib64/libpython3.6m.so.1.0+0x12b20f)
[... 54 lines cut ...]
  0x0c0c7fff95a0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c0c7fff95b0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c0c7fff95c0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
==13820==ABORTING

```

noc0lour, 2019-03-31:
This exact issue was resolved with d6e217cbc00ec0333ba030ab2d84c31b7dbb77d but it's unclear why.

marcusmueller, 2019-04-21:
Giving this another 24 days of clarification period. Please close after 2019-05-15 if not clarified.

noc0lour, 2019-04-23:
Well, access to a tag vector from within python via index is just not possible and results in out-of-bounds access, this has been addressed in the QA by not doing that. It seems this was always just pure luck that the memory access worked at all. Probably best if you put something in the docs that `.tags()[0]` is not supported and will results a coredump.

closed 2019-05-14 by noc0lour