gnuradio/gnuradio #2282: CMake: Convert from `find_package` + `set(GR_REQUIRED_COMPONENTS)` + `target_link_libraries${GNURADIO_FOO_LIBRARIES}` by modern CMake

marcusmueller, 2019-01-03:
As @noc0lour has prepared work (https://github.com/noc0lour/gnuradio/tree/update_cmake) to convert from the slightly backward way we let our users detect and link against GR components to modern CMake, where you just tell CMake to link against `gnuradio::digital`, we should be getting this out the door for 3.8.

*  makes things better debuggable
* reduces redundant code
* eases gr-modtool 3.8-porting

noc0lour, 2019-01-03:
Input from @michaelld very much appreciated.

michaelld, 2019-01-03:
Very cool! Is there a formal PR for this work yet? Should I comment directly in @noc0lour 's branch? I need to do actual testing / verification, but I like what I'm seeing during a brief read-through!

noc0lour, 2019-01-03:
Oh, I can post it as wip PR, comments in branches are very hard to work on.

michaelld, 2019-01-03:
I'd say to do the WIP PR then & we'll work on it via comments and some more commits.

noc0lour, 2019-01-03:
oh, it's already open. I just forgot: https://github.com/gnuradio/gnuradio/pull/2230

michaelld, 2019-01-04:
OK I've done a first read-through & commenting. I'll try out the PR sometime soonish to verify that the "modern CMake" stuff works. I love the code simplification & hierarchal nature of "modern CMake" and how you're using it here!

closed 2019-03-08 by noc0lour