gnuradio/gnuradio #2285: Test all flow graphs, make sure they're converted to YAML

mbr0wn, 2019-01-04:
Let's make sure we've gone over all the flow graphs. Please help update this. If you can't edit the first comment (i.e., this one), post below and I or someone else will update this.

### gr-uhd

- [ ] uhd_const_wave.grc
- [ ] uhd_fft.grc
- [ ] uhd_msg_tune.grc
- [ ] uhd_normalized_gain.grc
- [ ] uhd_siggen_gui.grc
- [ ] uhd_two_tone_loopback.grc
- [ ] uhd_wbfm_receive.grc

haakov, 2019-01-05:
I'm sure you've figured it out, but just in case anyone else is wondering what this "float division bug" is:

Python 2 returns an integer when dividing integers, whereas Python 3 returns a float. This causes expressions' types to change for some flowgraphs when opened in 3.8, and typically results in an error like "Expression X is invalid for type Y"

jdemel, 2019-01-05:
Exactly. In order to make it compatible across Py2k and Py3k, it is sufficient to add `//` instead of `/` for integer divisions. Also, this makes the command more verbose.

mbr0wn, 2019-01-05:
@jdemel Yes, that's exactly what I propose.

drmpeg, 2019-01-07:
I have all the DTV examples regenerated, but I'm waiting on PR https://github.com/gnuradio/gnuradio/pull/2258 to get the layouts nice and pretty.

jdemel, 2019-01-12:
**gr-analog** 

- [x] noise_power.grc

jdemel, 2019-01-19:
**gr-zeromq**

- [x] zeromq_pubsub.grc
- [x] zeromq_pushpull.grc
- [x] zeromq_reqrep.grc
- [x] zmq_msg.grc
- [x] zmq_stream.grc

jdemel, 2019-07-24:
Regarding the current state of this issue:
If you run `find . -type f -name "*.grc" -print0 | xargs -0 grep -in xml`, the result will be:

```
./grc/tests/resources/test_compiler.grc:1:<?xml version='1.0' encoding='utf-8'?>
./docs/exploring-gnuradio/fm_tx.grc:1:<?xml version='1.0' encoding='ASCII'?>
./docs/exploring-gnuradio/fm_rx.grc:1:<?xml version='1.0' encoding='ASCII'?>
./gr-trellis/examples/grc/tcm.grc:1:<?xml version='1.0' encoding='utf-8'?>
./gr-trellis/examples/grc/sccc.grc:1:<?xml version='1.0' encoding='utf-8'?>
./gr-trellis/examples/grc/turbo_equalization.grc:1:<?xml version='1.0' encoding='utf-8'?>
./gr-trellis/examples/grc/viterbi_equalization.grc:1:<?xml version='1.0' encoding='utf-8'?>
./gr-trellis/examples/grc/pccc.grc:1:<?xml version='1.0' encoding='utf-8'?>
./gr-trellis/examples/grc/interference_cancellation.grc:1:<?xml version='1.0' encoding='utf-8'?>
./gr-fec/examples/ber_curve_gen_ldpc.grc:1:<?xml version='1.0' encoding='utf-8'?>
./gr-fec/examples/fecapi_tagged_decoders.grc:1:<?xml version='1.0' encoding='utf-8'?>
./gr-fec/examples/ber_test.grc:1:<?xml version='1.0' encoding='utf-8'?>
./gr-fec/examples/tpc_ber_curve_gen.grc:1:<?xml version='1.0' encoding='ASCII'?>
```

- the `*_curve_gen.grc` examples are converted in #2615.
- `gr-fec/examples/fecapi_tagged_decoders.grc`: this `TypeError: string indices must be integers` needs to be fixed.
- `grc/tests/resources/test_compiler.grc` should not be converted. Or should it?
- the 2 files in `docs` should be converted.
- the `gr-trellis` examples need to be updated to only use 3.8 blocks and then should be converted.

marcusmueller, 2023-12-22:
current state, remaining XML FGs:

- gr-trellis/examples/grc/viterbi_equalization.grc
- gr-trellis/examples/grc/tcm.grc
- gr-trellis/examples/grc/sccc.grc
- gr-trellis/examples/grc/pccc.grc
- gr-trellis/examples/grc/turbo_equalization.grc
- gr-trellis/examples/grc/interference_cancellation.grc
- gr-fec/examples/ber_test.grc

Since opening them, running them, saving them and making a commit should be straightforward: tagging as good first issue

closed 2024-04-01 by willcode