gnuradio/gnuradio #2286: FM empthasis filter divide by zero condition always

tfcollins, 2019-01-04:
https://github.com/gnuradio/gnuradio/blob/d4225d6add2bda7e865e00aee31a57770336fa4b/gr-analog/python/analog/fm_emph.py#L292

Not sure if switching to python3 kicked this up, but I ran into it trying to do some FM TX examples. Will have to ask the original maintainer what that should be on line 291.

awalls-cx18, 2019-01-21:
I wrote the original.  That line originally called "cmath.rect(1.0, -w_0dB)" with 1.0 being the radius and -w_0dB being the angle.

https://docs.python.org/2/library/cmath.html#conversions-to-and-from-polar-coordinates
https://docs.python.org/3/library/cmath.html#conversions-to-and-from-polar-coordinates

The change to "cmath.rect(1.0 / -w_0dB)" is incorrect and the rest of the line got a bit mangled too.

git blame shows that this was part of a python2->python3 mega-change authored by @djanderson almost 2 years ago.

If you wait for this link to load completely, it will jump you to the change:

https://github.com/gnuradio/gnuradio/commit/9e625c4821f4c63421b3d3747c0c4f358fef6c5f#diff-5ad245c98ef1b80d3d8dbd14fd545247L295

The source line was mangled badly, likely due to some automated conversion script failing either due to multiple instance of "cmath.rect()" on the lines, or the line continuation escape character.

@marcusmueller and @djanderson : Someone should probably do a root cause analysis of why this automated change failed, and see if any other source lines could have also been transformed errantly.

marcusmueller, 2019-04-21:
Oh shoot, @awalls-cx18, how could I miss this?! Waaaah!

Ok, first of all, we need to fix this.

Then, we'll have to look through that patchset!

djanderson, 2019-04-21:
One of the problems with squashing commits... when I was working on this and used an automated tool I always committed just that changeset and added the exact command line invocation I used. Unfortunately I cleaned up my fork after this was merged.

You can see it was having an issue with the line continuation `\` because it changed lines 156 and 160 as well. 

I may have some time to scan the commit.

djanderson, 2019-04-21:
I will say that all tests were passing when this was merged so this must not be covered. Good chance to add a simple test that triggers this while we have the issue.

PR #2480 (2019-05-23): analog: correct errors of automatic conversion script in fm_preemph
An automatic conversion for python3 mangled the source and introduced a bug.
This commit fixes the conversion bug in fm_preemph. Changeset that lead to this
bug still has to be analyzed further.

Fixes #2286
commit 388ccbd8 (2019-05-23): analog: correct errors of automatic conversion script in fm_preemph

An automatic conversion for python3 mangled the source and introduced a bug.
This commit fixes the conversion bug in fm_preemph. Changeset that lead to this
bug still has to be analyzed further.

Fixes #2286
closed 2019-05-23 by mbr0wn